zwave-js/zwave-js #7128: No inclusion/exclusion, and some nodes becomed unknown after the first reboot (and remain unk after re-interviewing)

Vaufreyd, 2024-08-28:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.12.0.96eeb76

### ZwaveJS version

12.5.6

### Describe the bug

I have a raspberry pi5 with domoticz (on ubuntu). Updating from previous domoticz version, I also migrate from openzwave to zavejzui (running in docker). I encouter 2 problems:

- inclusion/exclusion do not work using button on devices. I can not exclude then, neither include new devices (inclusion using qrcode works fine)

- after migrating, everything went fine for all my devices (except dead nodes). After reboot, some of my fibaro wall plug are not recognize anymore (they are the same as the other ones). They are mark as unkown (90% of them). Even if the button switch on/off is seeable in zwavejsui, activating it do not work. The only part that is working seemlessly is the power consumption that works like a charm. As I can not exclude them/reinclude them, I can not use them anymore... Reinterview do not work. 

Thank you.

### To Reproduce

[zwave-js-ui-store.zip](https://github.com/user-attachments/files/16751011/zwave-js-ui-store.4.zip)

Tests concern node 082 (after time 2024-08-26T15:34:57.872Z) :
- first re-interview
- after exclusion attempt for this node
- after switch manipulation from this node (state change in zwavejsui but not on the device)

### Expected behavior

It was expected that :

- Inclusion/exclusion using button on device should work.
- Devices should not loose their manufacturer ID/ID.
- Reinterviewing nodes shloud solve the UNK problem.

### Additional context

Dongle Z‐Stick Gen5 USB Controller with a USB cable (1m) to put it away from computer. Usage on domoticz though MQTT discovery (previsouly openzwave) on PI5 on ubuntu 23.10. Zwavejsui deployed uising docker.

AlCalzone, 2024-08-29:
Not sure what happened, but I can't open that attachment.

AlCalzone, 2024-08-29:
When migrating from OZW, did you transfer the network key over to Z-Wave JS (called "S0 Legacy" now)? AFAIK OZW used security by default, which means most devices won't function until you tell Z-Wave JS the network key.

Also, there's a lot of traffic in the network. I wouldn't be surprised if some of the inclusion/exclusion issues are caused by that.
See also https://zwave-js.github.io/node-zwave-js/#/troubleshooting/first-steps?id=reduce-reports-from-the-network

closed 2024-09-16 by zwave-js-assistant[bot]