zwave-js/zwave-js #7130: IBT4ZWAVE used to work with Gen5 and not working with ZooZ stick

xabix, 2024-08-28:
Hello All,

The inclusion was done OK and the device is correctly added (and ping works and the interview too). But action like open or close don't work.

[image]

Is the device supported with the ZooZ stick, is it a pb of a firmware?

I see messages like:
- [Application Status CC (not implemented)]
- no handler for application command
- [Multilevel Switch] duration: "unknown" => "unknown"

Here is the log:
[Zwave.log](https://github.com/user-attachments/files/16784049/Zwave.log)
[Zwave-silly.log](https://github.com/user-attachments/files/16784119/Zwave-silly.log)

Thanks
XabiX

Version:
zwave-js-ui: 9.17.0zwave-js: 13.1.0home id: 3886169917home hex: 0xe7a23f3d

AlCalzone, 2024-08-29:
This is not an issue with the stick - it is "just" a radio for the most part. Issues with that device have been reported before, but AFAIK there has never been a solution or an answer what's going on.

After trying to control the device, it sends back an Application Status CC, with the Application Rejected Request command. This CC is not understood by Z-Wave JS, but typically means "can't do that". In this case the device actively responded that it won't execute the request.

Maybe the motor is jammed? Maybe it needs to be calibrated?

xabix, 2024-08-29:
Hi @AlCalzone 

What I don't understand is why he used to work with the same installation and Gen5 stick. If I understood you the good news is that if he used to work, he should still be working.

I did a manual calibration, and try to push for a level change to 40
[image]

Here is the log:
[ZM.log](https://github.com/user-attachments/files/16796373/ZM.log)

Should I look to make a level calibration directly from the Zwave/stick?

Thanks
XabiX

PS: https://www.niceforyou.com/sites/default/files/upload/manuals/idv01mm_en.pdf

AlCalzone, 2024-08-29:
Can you share a log of you controlling the device with the Gen5 stick then?

AlCalzone, 2024-08-29:
I see, that sounds like a hurdle. You could also contact the manufacturer and ask why the device responds with a **Application Status CC, with the Application Rejected Request command**.

Re-interviewing doesn't make sense  in this case.

xabix, 2024-09-02:
Do we know if the module is supposed to work? What I mean is the issue specific to me (could I have break/brick the device?) or all folks are having issues with this device?

It seems that it's working for some people, so I will search if this is linked to a device fw issue and yes will try to open a case with them then?
https://community.home-assistant.io/t/zwave-gate-module-nice-ibt4zwave/221803

it's supposed to be supported:
[OpenSmartHouse Z-Wave Device Database 29](https://opensmarthouse.org/zwavedatabase/1320)

AlCalzone, 2024-09-02:
It's working for some people. I can't tell you why yours isn't working, other than it refuses to react to controlling commands. Your best bet is to ask the manufacturer, they should be able to tell you.

mariusz-schimke, 2024-09-18:
If it may help in any way, I can add that it does work fine with my Zooz controller. The only two differences I can spot are:
* Zooz firmware version,
* My IBT4ZWAVE module couldn't be added securely.

Also, my network is on the EU radio bands, if it may matter (I noticed some differences in the FW version availability for Zooz based on the region).

[image]
[image]

xabix, 2025-07-30:
I did try with 1.5 and now 1.6 with EU bands (having done a re-interview) too but when I press 8-38-0 Start to open nothing happens.

[image]

Is there anything I can do to understand why it s not working?

[image]
[image]
[image]

xabix, 2025-07-31:
If I set target value to 99, the gate doesn't move. Is it the same issue or a different one that is not dependant to the transmitter/receiver?

```
2025-07-31 14:27:42.725 INFO Z-WAVE: Writing 99 to 8-38-0-targetValue
2025-07-31 14:27:42.725 CNTRLR   [Node 008] [setValue] calling SET_VALUE API MultilevelSwitchCCAPI:
                                   property:     targetValue
                                   property key: undefined
                                   optimistic:   true
2025-07-31 14:27:42.727 DRIVER   one or more queues busy
2025-07-31 14:27:42.728 DRIVER » [Node 008] [REQ] [SendDataBridge]
                                 │ source node id:   1
                                 │ transmit options: 0x25
                                 │ callback id:      89
                                 └─[Security2CCNonceGet]
                                     sequence number: 206
2025-07-31 14:27:42.729 SERIAL » 0x011100a900010008039f01ce25000000005961                            (19 bytes)
2025-07-31 14:27:42.733 SERIAL « [ACK]                                                                   (0x06)
2025-07-31 14:27:42.736 SERIAL « 0x010401a90152                                                       (6 bytes)
2025-07-31 14:27:42.736 SERIAL » [ACK]                                                                   (0x06)
2025-07-31 14:27:42.737 DRIVER « [RES] [SendDataBridge]
                                   was sent: true
2025-07-31 14:27:42.750 SERIAL « 0x011d00a95900000100a07f7f7f7f00000300000000030100007f7f7f7f7fcd    (31 bytes)
2025-07-31 14:27:42.750 SERIAL » [ACK]                                                                   (0x06)
2025-07-31 14:27:42.751 DRIVER « [REQ] [SendDataBridge]
                                   callback id:            89
                                   transmit status:        OK, took 10 ms
                                   routing attempts:       1
                                   protocol & route speed: Z-Wave, 100 kbit/s
                                   routing scheme:         LWR
                                   ACK RSSI:               -96 dBm
                                   ACK channel no.:        0
                                   TX channel no.:         0
2025-07-31 14:27:42.798 SERIAL « 0x012200a80000010008149f028201a682a221e0bfd77c9530b321d2414ad600a00 (36 bytes)
                                 07f7fbd
2025-07-31 14:27:42.799 SERIAL » [ACK]                                                                   (0x06)
2025-07-31 14:27:42.800 DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]
                                 │ RSSI: -96 dBm
                                 └─[Security2CCNonceReport]
                                     sequence number:  130
                                     SOS:              true
                                     MOS:              false
                                     receiver entropy: 0xa682a221e0bfd77c9530b321d2414ad6
2025-07-31 14:27:42.802 DRIVER » [Node 008] [REQ] [SendDataBridge]
                                 │ source node id:   1
                                 │ transmit options: 0x25
                                 │ callback id:      90
                                 └─[Security2CCMessageEncapsulation]
                                   │ sequence number: 207
                                   │ extensions:
                                   │ · type: SPAN
                                   │   sender EI: 0xa19d3663391bb77948345e98b36df6dc
                                   └─[MultilevelSwitchCCSet]
                                       target value: 99
2025-07-31 14:27:42.802 SERIAL » 0x013000a900010008229f03cf011241a19d3663391bb77948345e98b36df6dc288 (50 bytes)
                                 0ab6f5adc955acc97f53d25000000005a4e
2025-07-31 14:27:42.811 SERIAL « [ACK]                                                                   (0x06)
2025-07-31 14:27:42.813 SERIAL « 0x010401a90152                                                       (6 bytes)
2025-07-31 14:27:42.814 SERIAL » [ACK]                                                                   (0x06)
2025-07-31 14:27:42.814 DRIVER « [RES] [SendDataBridge]
                                   was sent: true
2025-07-31 14:27:42.860 SERIAL « 0x011d00a95a00000400a07f7f7f7f00000300000000030200007f7f7f7f7fc8    (31 bytes)
2025-07-31 14:27:42.861 SERIAL » [ACK]                                                                   (0x06)
2025-07-31 14:27:42.861 DRIVER « [REQ] [SendDataBridge]
                                   callback id:            90
                                   transmit status:        OK, took 40 ms
                                   routing attempts:       2
                                   protocol & route speed: Z-Wave, 100 kbit/s
                                   routing scheme:         LWR
                                   ACK RSSI:               -96 dBm
                                   ACK channel no.:        0
[... 100 lines cut ...]
2025-07-31 14:27:48.475 INFO Z-WAVE: [Node 008] Value updated: 38-0-duration unknown => unknown
2025-07-31 14:27:48.475 CNTRLR   [Node 008] [translateValueEvent: value updated]
                                   commandClass: Multilevel Switch
                                   endpoint:     0
                                   property:     targetValue
                                   propertyKey:  undefined
[... cut ...]
                                 │ RSSI: -96 dBm
                                 └─[Security2CCMessageEncapsulation]
                                   │ sequence number: 132
                                   │ security class:  S2_Authenticated
                                   └─[MultilevelSwitchCCReport]
                                       current value: unknown
                                       target value:  unknown
                                       duration:      unknown
```

AlCalzone, 2025-07-31:
Same issue. It's as if the device is in a state where it cannot move.

AlCalzone, 2025-07-31:
It's not a communication issue. The device receives the command just fine and then refuses to do it. I'm pretty sure it is something with the device, maybe even a physical issue.

closed 2025-08-04 by zwave-js-assistant[bot]