zwave-js/zwave-js #7131: Cannot set parameter Motion Sensor Sensitivity for device NAS-PD07Z

darkfuncat, 2024-08-28:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

NPM

### Z-Wave JS UI version

9.17.0

### ZwaveJS version

13.2.0

### Describe the bug

When I try to change parameter 5 "Motion Sensor Sensitivity " of device NAS-PD07Z I get "Unable to write ... : Fail" for any value other than 0 or 1. (theorically until 15 is good)

### To Reproduce

try to change the value other than 0 or 1 (theorically until 15 is good)

### Expected behavior

the value is changed without error

### Additional context

I suspect that I have a wrong config file, with a wrong parameter type.
My device is installed as device ID "0x0258 0x0010-0x0720" and I see there is another config file for the same device with a different ID "0x0258 0x0020-0x0720", perhaps I have the wrong config file ?
How can I change the device manufacturer ID in zwave-js-ui ?

AlCalzone, 2024-08-29:
It's entirely possible that the config file we have is wrong. Those Shenzhen Neo devices sometimes even ship with the wrong manual, so it doesn't get noticed until someone uses the paramters like you did.

Could you check whether the parameters mentioned in https://github.com/zwave-js/node-zwave-js/files/6758340/motion.5in1.manual.20200602.1.docx match your device? The manual is supposed to be for the 0x0020 variant, but maybe the parameters also fit for the 0x0010 one.

darkfuncat, 2024-08-29:
OK I made some tests, this is very odd.
I test field ID 5 named "Motion Sensor Sensitivity " in UI, but it interact exactly as the field "Binary Sensor Report Enable" in your documentation : value 0 OR 1, and device send actually binary report or not.
This would indicate that you are right the device is the 0x0020 variant.
BUT :
I also test the field ID 13 "Motion Clear Time" and it works as intended for the variant 0x0010. (i test 30 sec and 15 sec value and it actualy works)

At the end, I dont know, it seems a mix of 0x0010 and 0x0020 variant.
But the only parameter that interest me now is the "sensor sensitivity" and i cannot find it. (not 5 nor 6)
Perhaps another variant ?

AlCalzone, 2024-08-29:
Do you have a paper manual for that device by any chance?

darkfuncat, 2024-08-29:
This would be of course too easy and not fun =)
More seriously, no manual inside box.
Is it possible to change the variant ID manually ? Actually zwjsui detect it automatically as "0x0010" but it would be interesting to test it as "0x0020" variant. (I have 2 of these devices, both with same problem)

AlCalzone, 2024-08-29:
The solution would be to correct the config file for the 0x0010 variant, so you get offered the correct parameters for your device. Which is why I'm trying to figure out the correct parameters here.

I guess we can start by correcting parameter 5 to be Enable/Disable instead of the 0-15 range.

darkfuncat, 2025-02-11:
Hello, I give another try for this annoying issue.
I check on https://devices.zwave-js.io/?jumpTo=0x0258:0x0020:0x0718:0.0 and it seems that there are already 2 differents profil for the same device, Is it possible to work on a third profile ? or fix the second perhaps ?

[image]

Thanks