zwave-js/zwave-js #7165: Missing device configuration: Springs Window Fashions - CRZB

tommyjlong, 2024-09-13:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Springs Window Fashions

### Manufacturer ID

0x026e

### Product Type

0x5252

### Product ID

0x5a42

### Firmware Version

14.7

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

_Some pieces of information (could not find anything for this at Z-Wave Alliance):_
This is a Cellular Shade from Springs Windows Fashions (in my case branded as Bali).

The label on the shade is:
* Model: CRZB
* FCC-id: 2AAVX-CRZB (see [here](https://fcc.report/FCC-ID/2AAVX-CRZB))
(A manual [here ](https://fcc.report/FCC-ID/2AAVX-CRZB/7119970)starting p55 that covers both remotes and shades)

Description (from FCC-id): CRZB PowerCore Z-Wave with BLE Shade Technical Description

[Here ](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x026e/csz1.json) is a possible  existing config file for comparison.

lodged78, 2024-11-18:
I have 10 Bali blinds labeled CRZB. It appears this device was released in 2024, and is not yet in the Z-Wave Alliance database yet. I contacted the manufacturer for an update.

My CRZB blinds report two different product codes:
0x5252-0x5a42
0x4254-0x5a42

Eight report 0x5252, and two report 0x4254.

I would write a config file for CRZB, but I'm not sure how to manage two different product codes for one device.

tommyjlong, 2024-11-24:
I will add that I bought a second Bali Cellular Shade and it has the same label/codes as my original posting.
Label:
-  Model: CRZB
-  FCC-id: 2AAVX-CRZ

Codes:
-  Product Type: 0x5252
-  Product ID: 0x5a42

Also, still not seeing anything in the Z-Wave Alliance dB.

tommyjlong, 2024-11-25:
Here is the file that I came up with and [tested using ZWaveJS-UI](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files?id=using-custom-device-configuration-files). _I went into the ZwaveJS-UI HA Addon and stored this file in /data/store/config_.

filename: crzb.json
```
{
	"manufacturer": "Springs Window Fashions",
	"manufacturerId": "0x026e",
	"label": "CRZB",
	"description": "Cellular Shade Radio Z-Wave with Bluetooth",
	"devices": [
		{
			"productType": "0x5252",
			"productId": "0x5a42"
		},
		{
			"productType": "0x4254",
			"productId": "0x5a42"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"metadata": {
		"inclusion": "1. Put Primary Controller into Inclusion Mode\n\n2. Press and Hold Shade Programming Button for >3 seconds (LED flashes green, release the button)",
		"exclusion": "1. Put Primary Controller into Exclusion Mode\n\n2. Press and Hold Shade Programming Button for >3 seconds (LED flashes green, release the button)",
		"reset": "Please use this procedure only when the network primary controller is missing or otherwise inoperable\n\nPress and hold the button on the shade for approximately 7 seconds (shade jogs, release the button)",
		"manual": "https://support.springswf.net/helpdesk/attachments/30050113378"
	}
}

```
I don't have a yarn build environment to [run this through lint](https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files?id=process-to-submit-device-files).  All I did was to copy and edit the official csz1.json file and renamed it.

Also I don't have shades with the 0x4254-0x5a42 ids to test this.

lodged78, 2024-11-26:
I made and tested a config file last week and submitted a [pull request](https://github.com/zwave-js/node-zwave-js/pull/7416); I should have posted in this thread when I did it.

Essentially the same as your file, but I included notes on how to pair a z-wave hub _and_ a remote. Your manual link was better than the one I found that was split into two parts.

I paired some blinds with my hub and a BRZ1 remote, but later I tried to do the same with another set of blinds / remote but couldn't get it to work. I called the manufacturer, and they said it wasn't possible - but I knew it was because I had done it. I fiddled with pairing blinds with the hub and remote until I found a procedure that works every time. I believe my method adds the remote as a secondary controller, but I'm not certain.

```
{
	"manufacturer": "Springs Window Fashions",
	"manufacturerId": "0x026e",
	"label": "CRZB",
	"description": "Cellular Shade Radio Z-Wave",
	"devices": [
		{
			"productType": "0x4254",
			"productId": "0x5a42"
		},
		{
			"productType": "0x5252",
			"productId": "0x5a42"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"metadata": {
		"inclusion": "1. Put Primary Controller into Inclusion Mode\n\n2. Press and Hold Shade Programming Button for 3 seconds (release when LED flashes).\n\n\n\nTo include a shade and shade remote to the Z-Wave network, and pair the remote to the shade you must follow this sequence:\n\n1. Do a shade Z-Wave reset and then include into the Z-Wave network.\n\n2. Do a remote reset and then include the remote into the Z-Wave network.\n\n3. Pair the remote to the shade.",
		"exclusion": "1. Put Primary Controller into Exclusion Mode\n\n2. Press and Hold Shade Programming Button for 3 seconds (release when LED flashes).",
		"reset": "Z-Wave Reset:Exclude from the controller before doing this. Press and hold the button on the shade for approximately 10 seconds (release when the shade jogs, holding longer will perform a full reset).\n\nFull Reset: Please use this procedure only when the network primary controller is missing or otherwise inoperable. Press and hold the button on the shade for approximately 15 seconds (the LED will stop flashing when complete) - you will need to set top / bottom / home positions.",
		"manual": "https://support.springswf.net/helpdesk/attachments/30050113378"
	}
}
```

AlCalzone, 2025-03-11:
fixed in https://github.com/zwave-js/node-zwave-js/pull/7416

closed 2025-03-11 by AlCalzone