zwave-js/zwave-js #7192: Adding new devices Smart Scan Hit or Miss

heffneil, 2024-09-26:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

zwave-js-ui: 9.18.0.5040e7e 

### ZwaveJS version

13.2.0

### Describe the bug

I can not reliably add devices.  I just added a device and it worked. Then I follow the same procedure and it just never even tries to find it.  I have tried 4 or 5 new switches 10 times each.

### To Reproduce

On my Iphone with the latest updates (ios18) I go to provision and smart scan and scan the qr code on these zooz on/off switches. ZEN76 800LR

### Expected behavior

these switches should just get added.  I don't see inclusion mode activate on the top bar on my phone.  I have no idea what is going on but I am losing my mind trying to do this SO many times with poor to no results. Please help

AlCalzone, 2024-09-26:
Which Z-Wave stick do you have? On which firmware?

heffneil, 2024-09-26:
I am using a RaZberry 7 Pro with an external antenna on a network adapter from Tube

AlCalzone, 2024-09-26:
FWIW, dropping a 900000 lines logfile with the comment "more problems" is not particularly specific.
> They show in the list as unknown

Which nodes are "they"?

In any case, that logfile shows that you receive a meter report on average every 2.5 seconds, in some cases much more frequently. If routing is involved, there's likely even more permanent traffic on the mesh.
It's very likely that this has a lot to do with the issues you're seeing. I suggest giving https://zwave-js.github.io/node-zwave-js/#/troubleshooting/first-steps?id=reduce-reports-from-the-network a good read to improve that situation.

heffneil, 2024-09-26:
I will be honest I didn't look at the log files - I just zipped them up.  I tried to add several devices and played the join the network crapshoot game.  I was able to have 3 devices join quickly and properly and several others that took standing on one leg and patting the top of my head and rubbing my belly to get to join and others that just totally refused.  Now on day 2 I was able to get the last of them joined.  I appreciate the link.  I am struggling with over over-saturation of my wifi network and transitioning to zwave devices to clear up the airwaves.  Some growing pains getting the devices in but once they join and I stop adding more devices I think things are fairly reliable.

heffneil, 2024-09-26:
I see what you are saying and the problem is I looked on the zooz website about reporting and I don't know how to set it report less often.

Parameter 16: Choose how the switch reports its status back to the hub.

Values: 0 – Binary switch report for Z-Wave control, basic set report for paddle control; 1 – binary switch report for both Z-Wave and paddle control (default)

Size: 1 byte dec

AlCalzone, 2024-09-27:
> I will be honest I didn't look at the log files

What I meant to say, I need specifics to look for in such a huge logfile, e.g. _"Node xyz shows in the list as unknown and I can't seem to get it interviewed."_

> I don't know how to set it report less often.

Which specific device is _it_? Does it do energy metering?

heffneil, 2024-09-27:
I do have some lamp modules and other power monitoring devices but truthfully that's not what I am using them for.  99% are zooz devices ZEN04 800LR and 2 of the minoston  MP21Z / MP31Z

heffneil, 2024-10-08:
I have two nodes that are completely unknown to me.  They have been spinning for over a week now on the control panel page.  they do not appear in the smart start.  I have tried deleting failed nodes.  It still won't go away... Any thought or ideas?

heffneil, 2024-10-08:
Under Interviewee it says ProtocolInfo and its spinning - that's what i mean above. Restarting the container didn't fix it.

AlCalzone, 2024-10-08:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of you trying to remove the failed devices, loglevel `debug` and attach it here as a file (drag & drop into the text field).
Make sure to mention the node IDs, so I know what to look for.

heffneil, 2024-10-08:
I ran a rebuild routes just after sending this and 10 hours later 90% of them remain.  I am looking to invest in more Z Wave switches but I don't think I can consider it now.  Im just not sure what is wrong in my network.  I headed your advice and disable almost all power reporting except from 2 devices and when I actually can do stuff again I will disable it on them too. any and all advice is GREATLY appreciated.

heffneil, 2024-10-08:
```
2024-10-08T12:07:14.820Z CNTRLR   Serial API restarted unexpectedly.
2024-10-08T12:07:14.820Z CNTRLR   Currently active command will be retried...
2024-10-08T12:07:14.821Z CNTRLR   Starting hardware watchdog...
2024-10-08T12:07:15.079Z DRIVER   Dropping message with invalid payload
2024-10-08T12:07:15.110Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2024-10-08T12:07:15.242Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2024-10-08 08:07:15.245 INFO Z-WAVE: [Node 008] Metadata updated: 50-0-value-66049
2024-10-08 08:07:15.248 INFO Z-WAVE: [Node 008] Value updated: 50-0-value-66049 0 => 0
2024-10-08T12:07:15.274Z DRIVER   Dropping message with invalid payload
2024-10-08T12:07:15.450Z DRIVER   Dropping message with invalid payload
2024-10-08T12:07:15.515Z DRIVER   Dropping message with invalid payload```

AlCalzone, 2024-10-08:
@robertsLando does Z-Wave JS UI have an option to disable the watchdog yet?

robertsLando, 2024-10-09:
nope, is that a driver option?

AlCalzone, 2024-10-09:
Yes, added in v12.12.0:
https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#12120-2024-06-24

Looks like @heffneil needs to use that to fix their issues.

robertsLando, 2024-10-09:
Seems he can also use the env var `ZWAVEJS_DISABLE_WATCHDOG` for that, what do you think?

AlCalzone, 2024-10-09:
Depends on where Z-UI runs, but we should have a setting like disabling controller recovery.

robertsLando, 2024-10-09:
Created an issue on my side, in the meanwhile I think he can use the env var as he said he is using docker

heffneil, 2024-10-09:
Can I ask why I need controller recovery?

heffneil, 2024-10-09:
I am using a special configuration of Docker but I see I am running:

zwave-js-ui: 9.18.0.5040e7e
zwave-js: 13.2.0
home id: 4163067376
home hex: 0xf8235df0

robertsLando, 2024-10-17:
Option will be available on ui starting from next version 9.23.0+

closed 2024-10-17 by robertsLando