zwave-js/zwave-js #7200: Missing device configuration: <enter manufacturer and device name>

schford, 2024-09-28:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Aeotec TriSensor 8

### Manufacturer ID

881

### Product Type

0x0002

### Product ID

0x002d

### Firmware Version

FW: v2.7.1 SDK: v7.18.3

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/5016?selectedFrequencyId=1

### Do you have a manual?

[TriSensor-operation2024-1-3Rev9.0AeotecTriSensor.pdf](https://github.com/user-attachments/files/17174165/TriSensor-operation2024-1-3Rev9.0AeotecTriSensor.pdf)

schford, 2024-10-28:
Just realised I had wrong manufacturer ID have fixed it

AlCalzone, 2024-10-30:
@zwave-js-bot import config 5016 from zwa

PR #7342 (2024-10-30): feat(config): add Aeotec TriSensor 8
closes: #7337
fixes: #7200
closed 2024-10-30 by zwave-js-bot