zwave-js/zwave-js #7221: ZAC93 Zooz Controller/Zwave LR - no OTA firmware ever found

madasus, 2024-10-07:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.21.1.e8667de

### ZwaveJS version

13.4.0

### Describe the bug

I have a ZAC93 controller that I am using (it is behind an ESPHome device that provides access for ZwaveJS to connect to the ZAC93).  Everything in this setup is working well except that the OTA Updates page never shows any updates.  I know that the Zooz firmware updates are loaded into the repo so I would expect to see them.  

Example device

ZEN15 - i am running FW 2.0.1 and the latest is 2.30.2.  I have a bunch of these devices and I am able to manually load the firmware through ZwaveJS no problem.

My guess is that I am using Zwave LR as the protocol and perhaps that is somehow stopping the OTA version check.  Alternatively I've seen mentions of the region being incorrect but I'm not sure how to verify this.  I see "USA (Long Range)" as my region

### To Reproduce

1. Click on my out of date ZEN15
2. Go to the OTA Updates Tab
3. Hit Check Updates - "receive no updates available"

### Expected behavior

I believe it should show the correct Zooz firmware that needs to be loaded

### Additional context

ZAC93 is running on firmware 1.4 which is also the latest

madasus, 2024-10-07:
Oh - I had the same issue with the Zen32 800 LR which is there for sure.

I thought i had seen the ZEN15 in the firmware list but I must have been mistaken.

AlCalzone, 2024-10-08:
Can you share the device fingerprint and firmware version, or the device diagnostics of a device that should have an update but doesn't?

madasus, 2024-10-08:
I updated everything manually this time around.

Here is the device

0x027a 0x7000-0xa008

Its a Zen32 800LR

running 2.30.2 right now but was on 2.20.1 I believe before i updated manually.

Should have some more switches coming soon which will need updating if we need more info.

How can I add firmware to the repo? I have ZEN15 devices that need to be updated.

closed 2024-11-05 by AlCalzone