zwave-js/zwave-js #7223: Missing device configuration: Lockly GUARD DEADBOLT 728F-ZU

samrocketman, 2024-10-07:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

LOCKLY GUARD Z-Wave version smart lock

### Manufacturer ID

0x043D

### Product Type

0x03

### Product ID

0x0077

### Firmware Version

04.04.00

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4716

samrocketman, 2024-10-07:
[image]

# Technical manual

[image]

[image]

[image]

[image]

[image]

[image]

AlCalzone, 2024-10-30:
The link you shared seems to be for the PGD728F **ZPU**, not the PGD728F **ZU**. I can't find the latter on the ZWA page, so we have to go purely by the manual here.

samrocketman, 2024-10-30:
> The link you shared seems to be for the PGD728F **ZPU**, not the PGD728F **ZU**. I can't find the latter on the ZWA page, so we have to go purely by the manual here.

I noticed that and made a best effort “closest” link.  I reached out to the vendor for support and they emailed me a PDF so I will attach that

samrocketman, 2024-10-30:
Vendor-provided manual seems to match the pictures I took; I'll update the description.

[Zwave Technical Manual.pdf](https://github.com/user-attachments/files/17579883/Zwave.Technical.Manual.pdf)

PR #7382 (2025-03-11): feat(config): add device configurations for Lockly Secure Plus
Fixes: #7223 

The manual linked in the issue seems to indicate the same model name has two different device ids
closed 2025-03-11 by AlCalzone