zwave-js/zwave-js #7227: Secondary controller: Investigate if node ready events need to be emitted

AlCalzone, 2024-10-07:


eliasisrael, 2024-10-07:
Tell me if you need diagnostic data.

AlCalzone, 2024-10-08:
A [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the addon startup on loglevel `debug` would be good. Please attach it here as a file (drag & drop into the text field).

eliasisrael, 2024-10-10:
Hey, just so that you don't think I ghosted you...

I have tried to turn on logging to file, but it won't let me because the S2 and S0 keys are not in the config.

I'll attach a screen capture.

I imagine I can use the Silicon Labs software to extract the right keys from the device and then input those. I'll try that next.

[image]
[image]

AlCalzone, 2024-10-10:
Uhhh. How was it working before the update then? This means you didn't have secure communication before either, which seems weird for communication with an alarm panel.

AlCalzone, 2024-10-10:
Also @robertsLando having no keys isn't necessarily an error.

eliasisrael, 2024-10-10:
For all I know, it's a glitch in the display of that form, where information that is known to zwave-js is somehow not showing as far as the UI is concerned? Maybe it's because it's a secondary, and is using the keys provisioned for it by the primary?

I'm afraid that my ignorance of the zwave protocols and terminology prevents me from giving a complete answer.

eliasisrael, 2024-10-31:
OK. I apologize for the long delay, but I finally understand what was going on here.

I still had the z-wave JS add-on installed and I didn't realize that I should remove it, configure z-wave-js-ui, and then enable the INTEGRATION (not the add-on) using the z-wave-js server.

Now that I have done that, I see a much happier result.