zwave-js/zwave-js #7266: ZUI v.9.19.0: After a re-interview of node the "Multilevel Sensor v7" is no longer shown for QUBINO Flush 1/2 Relay nodes

MarkusElsing, 2024-10-10:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Snap

### Z-Wave JS UI version

V.9.19.0

### ZwaveJS version

V.9.19.0

### Describe the bug

Hello,
this is my first ticket for ZUI, I hope I submit this at the right place. The issue I have is that with the recent v.9.19  of it in my SNAP installation a re-interview of my nodes made the QUBINO Flush 1/2 Shutters (ZMNHBD and ZMNHAD, FW 5.0 and 6.1 respectively) lose the "Multilevel Sensor v7". This is for the temperature sensors that one can plug into the QUBINOs. I do have other nodes, like e.g. a Fibaro Smoke Detector (FGSD002) that shows successfully "Multilevel Sensor v8". With the previous version of the ZUI (v.9.17) the QUBINOs showed the v7 fine, so this is maybe something that recently came in with a change ?

Just to add, I use a RP4, openhab with MQTT and ZUI. I have 2 controllers, either a Aeotec Gen5 (FW 1.2) or a Z-Stick 7 (FW 7.20 and also an experimental 7.21 FW). The problem comes up with both controllers. I talked to the Aeotec support, they say this is not the controller FW, but ZUI that is probably not dealing correctly with the information from the nodes. They gave me their test FW to address some timeouts I observed with the 7.20 FW on the Z-Stick 7, which I first suspected to be the cause of the problem.

Many thanks in advance for your support.

Best, Markus Elsing

### To Reproduce

One needs a node with "Multilevel Sensor v7" (like the QUBINOs) and re-interview it. I have still seven nodes from QUBINO and all of them came up with the wrong configuration after a re-interview.

### Expected behavior

A re-interview should show the  "Multilevel Sensor v7" for the nodes.

AlCalzone, 2024-10-10:
@MarkusElsing the device does not report support for the multisensor. Also it does not expose the endpoint where it should exist. Typically this is the case when you've either changed some configuration exposing the endpoint, or attached the external sensor after including the device in the network.

This should solve it:
- Make sure the sensor is connected
- (If it exists,) make sure any configuration for Endpoint 3 is set correctly
- Exclude the device
- Include it again.

MarkusElsing, 2024-10-10:
Hi, thanks again for the fast feedback. I can try to exclude/include the device. Before the re-interview the device had the multisensor working fine, I was reading and histogramming the temperature in the openhab UI. So I didn't think this could be the problem. Maybe a loose connection or something. I'll test this and report back. I am traveling right now, will be Monday or so.

MarkusElsing, 2024-10-14:
Hi, after taking the temperature sensor out, excluding, putting it in and including, the node shows up indeed with multilevel sensor v7. But the reading is -999 deg, which looks like the sensor is broken or the node does not read it correctly. I also have issues with the first interview of the node, in the logfile I find:

2024-10-14T17:41:22.217Z CNTRLR   [Node 197] All attempts to assign lifeline group #8 failed, skipp [Endpoint 1]
                                  ing...
2024-10-14T17:42:01.475Z CNTRLR   [Node 197] All attempts to assign lifeline group #8 failed, skipp [Endpoint 2]
                                  ing...

I also have other odd things, like the node does not show correctly the endpoint 2 power meter in the report.

The odds are the node is broken and that started all the issues. I will replace it with another and see if this recovers it. I think this report can be closed as node issues are not for this forum. Thanks for the support !

closed 2024-10-14 by MarkusElsing