zwave-js/zwave-js #7267: All door sensors stopped working

Thaui57, 2024-10-10:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.21.1

### ZwaveJS version

13.4.0

### Describe the bug

After updating to 9.21.1 my door sensors do not respond anymore. In ZUI the open/close status will be indicated as 0/255 in Basic. In Domoticz the device has stopped working. So I desided to re-interview the device. I am wordering that the reinterviewed device is showing changes in the Basic V1 and I have no explantion why. I have compared with other door sensors which also has stopped working and the pictures are showing the difference between old status and re-interviewed status. 

### To Reproduce

[image]

### Expected behavior

[image]

AlCalzone, 2024-10-10:
In addition, which versions were you on before?

robertsLando, 2024-10-10:
Seems he was on 9.18

Thaui57, 2024-10-10:
after reflash door sensors are working again

robertsLando, 2024-10-10:
@AlCalzone Just to note that he is came back to 9.18

AlCalzone, 2024-10-10:
Oh then I misunderstood.

AlCalzone, 2024-10-10:
To allow me to investigate, please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
The log should include a re-interview of an affected device and an operation that's not working correctly.

Thaui57, 2024-10-10:
Can Not sende you a Driver log. I have flashed a 10 days old backup. All Sensors are Back and working.

closed 2024-10-10 by Thaui57