zwave-js/zwave-js #7276: Missing device configuration: Minoston MP26Z

CadenaMike, 2024-10-14:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Minoston MP26Z

### Manufacturer ID

0x0312

### Product Type

0xff01

### Product ID

0xff97

### Firmware Version

8.0.0

### Do you have a manual?

[MP26Z(800 Series) Manual.pdf](https://github.com/user-attachments/files/17359263/MP26Z.800.Series.Manual.pdf)

apella12, 2024-10-22:
You might want to look at this PR. The name is different, but the fingerprint is the same. #7302

benjaminflessner, 2026-08-15:
Is there a process for having multiple configs with the same fingerprint? I have a MP26Z and it, indeed, reports `786:65281:65431` (`x312:xff01:xff97`) just like the MP24Z. But, at least according to the linked manuals, there's a slight difference:

**[MP24Z](https://cdn-files.myshopline.com/file/store/1701829637050/MP24Z(800-Series)-Manual-V2-0-2023-12-7.pdf) (from https://github.com/zwave-js/zwave-js/pull/7302)**
> --Parameter =1, size =1 byte,
value=00 (default) LED is On when switch (light) is On
value=01 LED is On when the switch (light) is Off
value=02 LED is always Off

**[MP26Z](https://github.com/user-attachments/files/17359263/MP26Z.800.Series.Manual.pdf) (linked here)**
> Parameter =1, size =1byte, value=00(default) Led on when outlet on
Parameter =1, size =1byte, value=01 Led off when outlet on
Parameter =1, size =1byte, value=02 LED always ON
Parameter =1, size =1byte, value=03 LED always OFF

Talk about an annoying difference!

If we can't make a separate config, I don't suppose we have some sort of "pseudo setting" that could be used to differentiate the values / choose which variant of the config is used? Otherwise could/should we rename the existing config to MP24Z_MP26Z_800 and update the options to something horrible like:
2 - Always off (MP24Z) / on (MP26Z)
3 - Always off (MP26Z only)

Users can always manually set the value to 3 (which is what I did), but the point of configs is to make this easier, right?

apella12, 2026-08-16:
@benjaminflessner 
The firmware version can be used to separate the parameters. Something like this;
```
		{
			"#": "1",
			"$if": "firmwareVersion < 8.0",
		},
		{
			"#": "1",
			"$if": "firmwareVersion >= 8.0",
		},
```
The challenge would be if that is the correct breakpoint for the firmware. I don't see Minoston in the firmware repository. Odds are good the 800 series has a higher firmware level that the earlier version

benjaminflessner, 2026-08-18:
That's a thought. Though frustratingly, Minoston doesn't list any [firmware](https://minoston.com/pages/firmware-driver) for either model. My MP26Z (800) is reporting `8.20.0` in HA.

On the other hand - there already _is_ a config for the [MP24Z](https://github.com/zwave-js/zwave-js/blob/master/packages/config/config/devices/0x0312/mp24z.json) vs [MP24Z_800](https://github.com/zwave-js/zwave-js/blob/master/packages/config/config/devices/0x0312/mp24z_800.json). The problem is that BOTH devices being discussed here are the 800-series and yet report the same fingerprint:

[MP24Z (800) At Minoston](https://minoston.com/products/minoston-z-wave-800-series-dual-smart-plug-for-outdoor-and-indoor-mp24z)
[MP26Z (800) At Minoston](https://minoston.com/products/minoston-z-wave-800-series-dual-on-off-smart-plugmp26z)

I'm pinging them to get more information as to why they have 2 different devices with the same hardware id but different parameter behavior!

apella12, 2026-08-19:
From your links the devices look identical, except for one is outdoor and the other is indoor. Minoston does seem to reuse fingerprints, but the 800 versions have a higher firmware. For reference there is an example in the zwave-js code mp22zp.json and mp22zp_800. There are also two minoston templates `"templates/minoston_template.json#led_indicator_four_options_inverted"` and `"~/templates/master_template.json#led_indicator_three_options_inverted"` to cover both options