zwave-js/zwave-js #7277: Fibaro FGR-224 command cover.open cover.up not working

Loic691, 2024-10-14:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

last one

### ZwaveJS version

last one

### Describe the bug

After 15 or 30 days randomly, commands cover.open and cover.up are not working from the HA zwave integration
The set set_cover_position works perfectly.
When I have the issues, I can close or open my cover with set_cover_position.
I tested in zwave JS UI and the command works
[image]
Cover calibration is OK

EDIT : I saw open and closed are inverted after few time....
When the cover is closed (0%) I can click on the UP arrow but nothing happens
If I call the actions cover.open=> nothing happens
If I call the actions cover.close => The cover opens.... => STRANGE

If I manually set the position to 100%, the cover is opening => OK
When the cover is opened, if I click on the Down arrow, nothing happens
If I call the actions cover.closen=> nothing happens
If I call the actions cover.open => The cover opens.... => STRANGE

The command are also inverted into ZWAVE-JS UI
[image]

### To Reproduce

Difficult to reproduce.
The issues appears between 15 or 30 days.
When I reboot the Fibaro FGR-224, the issues is no present
I don't know if it's a Fibaro bug or ZWave JS UI bug...

### Expected behavior

Commands works

closed 2024-10-14 by zwave-js-bot

AlCalzone, 2024-10-14:
The bug is in `node-zwave-js` which both HA and Z-UI use. v13.9.0 fixes the issue, so you'll have to wait for a new release of Z-UI and/or the addon.

MUN0X, 2025-09-06:
Was this issue resolved?

I’m about to buy 9 FGR-224 modules but wondering if i should go for 223 instead.