zwave-js/zwave-js #7320: Add/remove associations not possible on the Heatit z-trm2 TF021 thermostat

AlCalzone, 2024-10-25:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/7262

<div type='discussions-op-text'>

<sup>Originally posted by **@nesheimt** October  9, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [x] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
1) I'm not able to add or remove associations
2) Relay state is not updated
3) Temperature reports are only comming every 9 hours (and I have set reg 17 to 6, which is report every minute)

What did you expect to happen instead?
I'd expect to add/remove associations and the relay state and temperature updates. 

Steps to reproduce the behavior:
I'm mainly using the zwave js ui front end to change associations, and look for updates in relay state and temperature. The result is identical if I use the Homeseer menu funcions, no difference that is.

Anything else we should know?
The TF021 has updated firmware v1.92 I can do assiciations setuop on other devices without any problems. I see that there are similar open/on-hold discussions/issues but they have no activity, but was asked by AlCalzone to threate this discussion. Main problem, associaitons, are not listed anywhere.
I think that root cause of my issues are problems with associations and if solved relay state and temperature reports might be fine. I can also ask the manufacturer helpdesk for information if required(I saw this was required in another issue), but the I need detailed instructions.

[HeatIt_1v9.pdf](https://github.com/user-attachments/files/17310750/HeatIt_1v9.pdf)

### Software versions

zwave-js-ui: 9.20.0
zwave-js: 13.3.1
home id: 3699079014
home hex: 0xdc7b7766

I'm using Homeseer4 v4.2.21.0 system, with the Z-wave plus 1.0.2.0 plug-in.

### Z-Wave Controller (Stick/Dongle/...)

Homeseer Smartstick G8

### Device information

Manufacturer: Heatit 0x019b
Model name: TF021 Thermostat  0x0003-0x0201
Node ID: 42

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [X] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

The logfiles contain a Re-interview and trying to remove Group 2 and 3

[zwave-js-ui-store (1).zip](https://github.com/user-attachments/files/17311040/zwave-js-ui-store.1.zip)
[node_42_dump (1).json](https://github.com/user-attachments/files/17311056/node_42_dump.1.json)
[node_42 (1).json](https://github.com/user-attachments/files/17311057/node_42.1.json)
[nodes_dump.json](https://github.com/user-attachments/files/17311070/nodes_dump.json)

</div>

AlCalzone, 2024-10-25:
@nesheimt Did the same problem exist on firmwares older than 1.92? Or is this a new bug introduced by that firmware?

nesheimt, 2024-10-25:
Previously I used the 1.92-device with Homeseer and their own plug-in for z-wave, and everything worked fine. I think I did the fw-upfate back in 2020 and experienced no issues thereafter. The problems were introduced when I migrated to their new "z-wave plus plug-in" based on z-wave js back i august this year.

The purpose with the v1.92 firmware was to fix relay and temperature reporting that was flawed in the previous firmwares. In 1.92 they "moved" temperature and relay reporting to association groups(lifeline, relay, temp1, temp2, temp3), and introduced two new registers(17 and 18) to control the temperature report rate(previously flooding the network)

To me it seems like the association group issues is the main problem(I cannot add or remove any associations), is related to the reporting issues(since they are based on associations groups).

I asked the manufacturer for documentation for the v1.92 and received the attached datasheet(even though it says v1.9)

[Appendix_Firmware_update_1_9_Heatit_Z-Wave.pdf](https://github.com/user-attachments/files/17517233/Appendix_Firmware_update_1_9_Heatit_Z-Wave.pdf)

AlCalzone, 2024-10-25:
> Previously I used the 1.92-device with Homeseer and their own plug-in for z-wave, and everything worked fine

Okay then you weren't affected by that firmware bug, since effectively no other driver uses Supervision by default. I'll add the workaround for the next release, then you should at least be able to manage the associations. Once that works, we can figure out if we should automatically some other associations than group 1.