zwave-js/zwave-js #7323: Ota firmware not found

spacemanspiff2007, 2024-10-25:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.24.0.5531846

### ZwaveJS version

13.9.1

### Describe the bug

OTA updates shows no firmware available.

[image]

Device has 10.27, available is 10.30.
[Firmware is there](https://github.com/zwave-js/firmware-updates/blob/d6f781569543597c42cee014d218d7f238798388/firmwares/shelly/qnsw-002P16.json#L17)

### To Reproduce

Open OTA Updates, nothing there

### Expected behavior

Ota update is available

AlCalzone, 2024-10-25:
I'm going to need some more details:
- Which controller are you using? Which firmware is it on?
- Which RF region is it set to? Please make a screenshot of the UI where you have the details for the controller open.
- Please make an [application log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=application-logs), not driver log, where you click the button to search for updates. Attach it here as a file (drag & drop into the text field).

spacemanspiff2007, 2024-10-26:
RF Region:
I changed if from Europe (default) to Europe when trying the fix ota update
[image]

This is the controller:
[image]

I manually upgraded the node to 10.30 because I was having issues with the device. I'll try to downgrade and reproduce the issue.

spacemanspiff2007, 2024-10-26:
Downgraded - no update
[image]

Had to restart because after settting level to debug the connection to the controller was lost.
Here is the log: [z-ui_current.log](https://github.com/user-attachments/files/17530292/z-ui_current.log)

AlCalzone, 2024-10-26:
That controller doesn't let us determine the region automatically. The setting has no effect because it also doesn't let us change the region.

Because of that, region specific firmware updates aren't offered, because we cannot make sure they don't brick your device.

There's another open issue to let the user manually specify the region in this situation, which would solve your problem.

spacemanspiff2007, 2024-10-26:
> The setting has no effect because it also doesn't let us change the region.

> here's another open issue to let the user manually specify the region in this situation

Can you explain why there are two region settings needed? 
Is it possible to use the controller in e.g. USA mode with EU devices?

Let's assume I have a controller where I can change the region and the setting has an effect:
When I change the region to USA I get an OTA download for USA and when I change it back to EU I would get an EU OTA download?
How would this be different than just selecting e.g. EU when the controller does not support specifying the region?

None the less I think there should be some kind of message that OTA updates are not possible.

AlCalzone, 2024-10-27:
> Let's assume I have a controller where I can change the region and the setting has an effect:
> When I change the region to USA I get an OTA download for USA and when I change it back to EU I would [...]

If we can query the region from the controller, we **know** it is set to that region and can only communicate with devices from that region. So if you are in USA, and set the controller to region EU, Z-Wave JS will download OTA updates for the EU region. But you won't be able to install them on your US devices, since the controller cannot perform the update on them.

If the controller does not support this, we **cannot know** which region it is in. So if the user is allowed to specify the region, we have to trust them that they specify the correct region. In this case, if you are in USA, and set the controller to region EU, Z-Wave JS will download OTA updates for the EU region. But this time it will be able to install that update on your US devices, since the controller is actually using the US region, not EU. Which means you can end up with bricked devices by selecting the wrong region.

closed 2024-11-05 by zwave-js-bot