zwave-js/zwave-js #7330: EVERSPRING SE813 Indoor Voice Siren: Unable to play sounds

AlCalzone, 2024-10-29:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/7324

<div type='discussions-op-text'>

<sup>Originally posted by **@damsellem** October 26, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
Inclusion is ok
Siren sounds well too
But i'm unable to make the siren speaks (but it worked very well with another older solution)

What did you expect to happen instead?
I except the siren to speaks recorded sounds

Steps to reproduce the behavior:
  1. In zwave js ui, Go to product list
  2. Select everspring siren 
  3. Go to Sound Switch V1 section
  4. Select a ToneId (except 0)
  5. Click on Play 
  6. An error is logged "2024-10-26 18:07:54.530 ERROR Z-WAVE: Unable to write 19 on 121-0-toneId: Fail"

Anything else we should know?

### Software versions

zwave-js-ui: 9.12.0.96eeb76
zwave-js: 12.9.1
home id: 3829344995
home hex: 0xe43f2ae3

If you are using something non-standard, tell us here: 
Zwave-js UI use with jeedom solution, but I reproduced the issue with homeassistant too

### Z-Wave Controller (Stick/Dongle/...)

AEON Labs | Z‐Stick Gen5 USB Controller | ZW090

### Device information

Manufacturer: 0x0060
Model name: 0x000c:0x0003

Node ID: 63

https://devices.zwave-js.io/?jumpTo=0x0060:0x000c:0x0003:0.0

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [ ] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

[zwave-js-ui-store (1).zip](https://github.com/user-attachments/files/17531327/zwave-js-ui-store.1.zip)
</div>

AlCalzone, 2024-10-29:
The siren responds with a status of "Fail" to supervision-encapsulated requests

AlCalzone, 2024-10-29:
Fix is included in v13.10.3

damsellem, 2024-10-29:
I updated the config file manually 
The device works well now
Thanks a lot for your work !

PR #7331 (2024-10-29): fix(config): disable Supervision for Everspring SE813
fixes: #7330