zwave-js/zwave-js #7355: Sirens no longer detected in 9.26.0 (upgrading from 9.23.0)

Altycoder, 2024-11-04:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.26.0.47f4c89

### ZwaveJS version

13.10.3

### Describe the bug

I have 3 sirens that are no longer detected when upgrading to 9.26.0: 2x Aeon Labs Siren Gen 5 and one Popp Solar Outdoor Siren 2.

### To Reproduce

Upgraded container from 9.23.0 to 9.26.0

### Expected behavior

Sirens are detected as they were in 9.23.0

### Additional context

These 3 sirens were working fine in 9.23.0.  I've tried rolling back to 9.23.0 from 9.26.0 but this doesn't fix the problem so the change appears to be permanent.  Normally rolling back the container reverts any issues but not in this case.  

One of these sirens is only accessible via a ladder outside so I desperately need a software fix please!  Log attached.

[zwavejs_current.log](https://github.com/user-attachments/files/17615559/zwavejs_current.log)

AlCalzone, 2024-11-04:
Looking at the sheer amount of repeated traffic coming in, your mesh seems unhealthy.
Make sure to follow https://zwave-js.github.io/node-zwave-js/#/troubleshooting/first-steps

Altycoder, 2024-11-04:
Thanks, this mesh has been set up for some time, but I've only noticed issues recently with the 9.23 -> 9.26 upgrade which caused me a bit of pain.

I'll see if I can re-build the mesh and/or go through and turn a bunch of unnecessary stuff off.

closed 2024-11-22 by zwave-js-assistant[bot]