zwave-js/zwave-js #7385: nvmedit package doesn't recognize backup made from EZZee 500 series controller

zwolfpack, 2024-11-11:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Manually built (git clone - npm run build - npm install

### Z-Wave JS UI version

9.26.0

### ZwaveJS version

13.10.3

### Describe the bug

I have an EZZee 500 controller from Express Controls, Inc. FW is v2.8, SDK v6.8.0.
https://devices.zwave-js.io/?jumpTo=0x001e:0x0005:0x0002:0.0

The nvmedit package doesn't recognize it as valid. 

### To Reproduce

From UI, Advanced actions->NVM Management->Restore, select and submit backup file
or from command line, nvmedit.js 'convert' or 'nvm2json' operations. 

Result from 'convert' operation is message 
```
ZWaveError: Did not find a matching NVM 500 parser implementation! Make sure that the NVM data belongs to a controller with Z-Wave SDK 6.61 or higher. (ZW0280)
    at NVM500.init (/opt/zwave-js/node_modules/@zwave-js/nvmedit/build/lib/NVM500.js:58:19)
    at async nvm500ToJSON (/opt/zwave-js/node_modules/@zwave-js/nvmedit/build/convert.js:604:18)
    at async migrateNVM (/opt/zwave-js/node_modules/@zwave-js/nvmedit/build/convert.js:1289:23)
    at async Object.handler (/opt/zwave-js/node_modules/@zwave-js/nvmedit/build/cli.js:168:20) {
  code: 280,
  context: undefined,
  transactionSource: undefined
}

```

### Expected behavior

From UI, successful restore to the same or different controller. From command line, successful conversion to binary or json output. 

### Additional context

I've dug into the nvmedit code and determined that it matches the Static_6_7x map and it's protocolVersion is 6.01. This protocolVersion is listed in the Static_6_8x map instead. 

Patching build/lib/nvm500/impls/Static_6_7x.js, adding 6.01 to protocolVersions list allows the conversion to proceed further. It then fails with "NVM data for property ... not found" for four parameters that are absent in 6_7x vs. 6_8x. Patching build/convert.js, setting the adapter.get 'required' parameter to 'false' for these parameters allows the conversion to complete successfully. 

Unless I missed something, these parameters being required will be an issue for any 6_7x or 6_6x NVM. Parameters absent in 6_7x are "systemState", "powerMode", "powerModeExtintEnable", "powerModeWutTimeout". Additional parameters absent in 6_6x are "powerLevelNormal" and "powerLevelLow". 

Is it possible to accommodate this controller? I realize there would be issues if there are 6.01 controllers out there that match the 6_8x map. With my patches in place, a '... json2nvm --protocolVersion 6.01 ...' operation produces a 6_7x mapping vs. a 6_8x mapping without the patches. 

FYI, EZZee of various revisions are used in the HomeSeer 'Z-Net' network enabled controllers

NVM backup file provided if requested. 

Thanks

AlCalzone, 2024-11-11:
Is that SDK 6.8.0 or 6.80.0?

zwolfpack, 2024-11-11:
Sorry, it’s 6.81.0.

zwolfpack, 2024-11-11:
Here you go
[zwave-js-ui-store.zip](https://github.com/user-attachments/files/17706990/zwave-js-ui-store.zip)

AlCalzone, 2024-11-13:
I've tried to find a hint in the atrocious documentation (and some terrible C code) I have on this stuff, but there is no clear indication in which SDK/Protocol version the layout was changed. It's entirely possible that SDK 6.81.1 (protocol 6.02) is the first to switch to the 6.8x layout and SDK 6.81.0 still uses the 6.7x format.

AlCalzone, 2024-11-13:
So that NVM seems to have some odd format where the end of the NVM is in a different location than for most 6.7x NVMs.
With the linked PR, Z-Wave JS can now parse it, but I'm not sure if the controller will be able to work when restoring a different NVM onto it.

closed 2024-11-13 by zwave-js-bot

zwolfpack, 2024-11-13:
Great, thanks!
I noticed that extra space, 303 byte mystery offset in the "Application" module. I tried loading a file without the offset; the controller appeared to work; however the backup made from that had corruption in the next module and wouldn't pass the nvmedit parsing criteria.

PR #7399 (2024-11-13): fix(nvmedit): parsing of 6.6x and 6.7x NVMs, use 6.7x NVM parser for protocol 6.01
fixes: #7385