zwave-js/zwave-js #7398: Cannot initialize controller, but I didn't make any changes, the /dev/serial/by-id shows the same path as before...

Majestic7979, 2024-11-13:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [ ] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

See bug description

### ZwaveJS version

See bug description

### Describe the bug

zwave-js-ui: 9.27.0.f3c4ac5
zwave-js: 14.3.1
home id: undefined
home hex: undefined

All was working at least until yesterday morning when I last used the door from outside, my door lock is z-wave, that's how I know it was fine at least up to yesterday. Today my housemate was locked out, door lock would not respond, I checked zwavejsui and it had this error:
```
Driver: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200) at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:6071:23) at ZWaveController2.queryCapabilities (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:1144:37) at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1677:46) at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1478:16) (ZW0100)
```
I navigated to /dev/serial/by-id and the stick is recognized. I checked my docker CLI to ensure it was fine, and it was. I have my container auto-update itself via Watchtower, so the only thing that may have changed from yesterday to today is the zwavejsui version, and I believe this is the issue because upon logging in I received the usual "what's new" popup.

So whatever you did on the latest update as of this post, broke my Aeotec stick. When I say broke, don't worry I don't mean you killed it. I'm just saying it is not working. I will now try to use image tag to downgrade and see if it works again, will update this if it does. Otherwise please be aware it's non-functional and thankfully I was inside to open the door otherwise we'd have no way to get in and would have to break the door! So please treat this urgently as it may affect others, for instance it would be catastrophic if there was a fire and their lock was not operational because the latest update fubar'd something.

Thanks for reading my report, I appreciate your effort in making this software available!

### To Reproduce

Not applicable

### Expected behavior

Not break the driver...

AlCalzone, 2024-11-13:
If this is still happening with 9.27.1, I definitely need to see driver logs.

Side note: IMO it seems a bit negligent to auto-update critical infrastructure, e.g. when doors cannot be opened without it working

AlCalzone, 2024-11-13:
Just tested again, with a fresh pull of 9.27.1. Works fine for me, both with a 500 series and a 800 series USB controller.

willheineman, 2024-11-14:
Also working again with a fresh pull of 9.27.0

AlCalzone, 2024-11-14:
@Majestic7979 can you double check if pulling 9.27.2 (latest version) solves your issue?

willheineman, 2024-11-14:
Confirmed that 9.27.2 is also working for me, but using the `latest` tag still defaults back to 9.27.0 (also working still)

AlCalzone, 2024-11-14:
@robertsLando Any idea what's going on with the tag?

robertsLando, 2024-11-14:
I checked on https://hub.docker.com/r/zwavejs/zwave-js-ui/tags and the shas of 9.27.2 match latest so sincerely I have no clue

disconn3ct, 2024-11-14:
I'm still getting this with a fresh 9.27.2 pull:
`zwavejs2mqtt-0 zwavejs2mqtt 2024-11-14 13:23:52.037 INFO STORE: Controller status: Driver: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)` My controller is via tcp (ser2sock) but it is working fine. 

A downgrade to 9.26.0 restores it immediately.

disconn3ct, 2024-11-15:
I've set the driver to debug and restarted, but it did not add the driver logs. I had to point it at a file before I could get anything:
``` driver.log
                                    ███╔╝  █████╗ ██║ █╗ ██║ ███████║ ██║   ██║ █████╗            ██║ ███████╗
                                   ███╔╝   ╚════╝ ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝       ██   ██║ ╚════██║
                                  ███████╗        ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗     ╚█████╔╝ ███████║
                                  ╚══════╝         ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝      ╚════╝  ╚══════╝
2024-11-15T12:53:49.223Z DRIVER   version 14.3.2
2024-11-15T12:53:49.224Z DRIVER   
2024-11-15T12:53:49.225Z DRIVER   starting driver...
2024-11-15T12:53:49.230Z DRIVER   opening serial port tcp://ser2sock-zwave:10000
2024-11-15T12:53:49.262Z DRIVER   serial port opened
2024-11-15T12:53:49.265Z SERIAL » 0x15                                                                 (1 bytes)
2024-11-15T12:53:49.772Z CNTRLR   [BOOTLOADER] !SER2SOCK Connected
                                  !SER2SOCK SERIAL_CONNECTED
2024-11-15T12:53:50.276Z DRIVER   loading configuration...
2024-11-15T12:53:50.285Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2024-11-15T12:53:50.287Z CONFIG   version 14.3.2
2024-11-15T12:53:51.919Z DRIVER   beginning interview...
2024-11-15T12:53:51.921Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2024-11-15T12:53:51.921Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2024-11-15T12:53:51.922Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2024-11-15T12:53:51.923Z DRIVER   added request handler for SetLearnMode (0x50)...
                                  1 registered
2024-11-15T12:53:51.924Z CNTRLR   querying Serial API capabilities...
2024-11-15T12:53:51.938Z SERIAL » 0x01030007fb                                                         (5 bytes)
2024-11-15T12:53:51.939Z DRIVER » [REQ] [GetSerialApiCapabilities]
2024-11-15T12:53:51.947Z SERIAL « +
2024-11-15T12:53:51.948Z CNTRLR   [BOOTLOADER] +
2024-11-15T12:53:51.949Z SERIAL « �
2024-11-15T12:53:51.950Z CNTRLR   [BOOTLOADER] �
2024-11-15T12:53:51.950Z SERIAL « 
2024-11-15T12:53:51.951Z CNTRLR   [BOOTLOADER] 
2024-11-15T12:53:51.952Z SERIAL « Z�����O����g
2024-11-15T12:53:51.953Z CNTRLR   [BOOTLOADER] Z�����O����g
2024-11-15T12:53:51.954Z SERIAL « ��
2024-11-15T12:53:51.955Z CNTRLR   [BOOTLOADER] ��
2024-11-15T12:53:51.955Z SERIAL « ��
2024-11-15T12:53:51.956Z CNTRLR   [BOOTLOADER] ��
2024-11-15T12:53:51.956Z SERIAL « 
2024-11-15T12:53:51.957Z CNTRLR   [BOOTLOADER] 
2024-11-15T12:53:51.958Z SERIAL « �s
2024-11-15T12:53:51.958Z CNTRLR   [BOOTLOADER] �s
2024-11-15T12:53:51.959Z SERIAL « �
2024-11-15T12:53:51.959Z CNTRLR   [BOOTLOADER] �
2024-11-15T12:53:51.960Z SERIAL « `
2024-11-15T12:53:51.961Z CNTRLR   [BOOTLOADER] `
2024-11-15T12:53:52.462Z CNTRLR   [BOOTLOADER] �
2024-11-15T12:53:52.942Z SERIAL « +
2024-11-15T12:53:52.944Z CNTRLR   [BOOTLOADER] +
2024-11-15T12:53:52.945Z SERIAL « �
2024-11-15T12:53:52.946Z CNTRLR   [BOOTLOADER] �
2024-11-15T12:53:52.947Z SERIAL « 
2024-11-15T12:53:52.949Z CNTRLR   [BOOTLOADER] 
2024-11-15T12:53:52.955Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2024-11-15T12:53:52.960Z SERIAL « Z�����O����g
2024-11-15T12:53:52.962Z CNTRLR   [BOOTLOADER] Z�����O����g
2024-11-15T12:53:52.963Z SERIAL « ��
2024-11-15T12:53:52.964Z CNTRLR   [BOOTLOADER] ��
2024-11-15T12:53:52.964Z SERIAL « ��
2024-11-15T12:53:52.965Z CNTRLR   [BOOTLOADER] ��
2024-11-15T12:53:52.966Z SERIAL « 
2024-11-15T12:53:52.967Z CNTRLR   [BOOTLOADER] 
2024-11-15T12:53:52.968Z SERIAL « �s
2024-11-15T12:53:52.969Z CNTRLR   [BOOTLOADER] �s
2024-11-15T12:53:52.970Z SERIAL « �
2024-11-15T12:53:52.971Z CNTRLR   [BOOTLOADER] �
[... 115 lines cut ...]
zwavejs2mqtt-0 zwavejs2mqtt     /usr/src/app/store/logs/zwavejs_2024-11-15.log
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:04.893 INFO STORE: Restarting client in 4 seconds, retry 3
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:08.899 INFO STORE: Client closed
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:08.920 INFO STORE: Connecting to tcp://ser2sock-zwave:10000
zwavejs2mqtt-0 zwavejs2mqtt Logging to file:
zwavejs2mqtt-0 zwavejs2mqtt     /usr/src/app/store/logs/zwavejs_2024-11-15.log
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:10.954 INFO STORE: GET /metrics 200 11.322 ms - 1
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:15.621 INFO STORE: Restarting client in 8 seconds, retry 4
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:23.628 INFO STORE: Client closed
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:23.638 INFO STORE: Connecting to tcp://ser2sock-zwave:10000
zwavejs2mqtt-0 zwavejs2mqtt Logging to file:
zwavejs2mqtt-0 zwavejs2mqtt     /usr/src/app/store/logs/zwavejs_2024-11-15.log
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:30.413 INFO STORE: Restarting client in 15 seconds, retry 5
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:45.416 INFO STORE: Client closed
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:54:45.440 INFO STORE: Connecting to tcp://ser2sock-zwave:10000
zwavejs2mqtt-0 zwavejs2mqtt Logging to file:
zwavejs2mqtt-0 zwavejs2mqtt     /usr/src/app/store/logs/zwavejs_2024-11-15.log
[... cut ...]
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:55:07.120 INFO STORE: Client closed
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:55:07.136 INFO STORE: Connecting to tcp://ser2sock-zwave:10000
zwavejs2mqtt-0 zwavejs2mqtt Logging to file:
zwavejs2mqtt-0 zwavejs2mqtt     /usr/src/app/store/logs/zwavejs_2024-11-15.log
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:55:10.717 INFO STORE: GET /metrics 200 5.343 ms - 1
zwavejs2mqtt-0 zwavejs2mqtt 2024-11-15 07:55:13.777 INFO STORE: Restarting client in 15 seconds, retry 7
```

AlCalzone, 2024-11-15:
```
2024-11-15T12:53:49.772Z CNTRLR   [BOOTLOADER] !SER2SOCK Connected
                                  !SER2SOCK SERIAL_CONNECTED
```
`ser2sock` is emitting ASCII text that causes Z-Wave JS to continue parsing the input as if it were in bootloader mode.

I think you need to run in raw device mode (`-0` flag on the command line) to disable that.

disconn3ct, 2024-11-15:
That was it! Thanks!

If anyone else hits this with ser2sock, the docker container entrypoint is a mess. (Even on my fork.) The fix is to force `command: /usr/local/bin/ser2sock` and then provide `args:`.

```yaml
command: /usr/local/bin/ser2sock
args: ["-0","-s","/dev/ttyUSB0","-b","115200"]
```

closed 2024-11-17 by AlCalzone

Majestic7979, 2024-11-19:
> @Majestic7979 can you double check if pulling 9.27.2 (latest version) solves your issue?

I am hesitant to do this due to recovering from a health issue at the moment and must have the ability to unlock the door at all times for emergency purposes, so will leave on the old tag 9.26 for now and will close the issue once I've had a chance to test. But I am scared of using zwavejs as my zwave network manager now because the Aeotec stick is probably the most commonly used stick out there and I sincerely find it shocking that something was distributed without checking that the stick driver is loaded correctly.

This could have caused a critical emergency situation in my case, I am not over-reacting just to make sure that my comment is understood. If a door lock (Conexis L2/L1) that is purely eletronic and has no key failsafe to open cannot be unlocked because ZwaveJS has a fault, it could mean a death situation.

I implore the developers to please test with the most common sticks before releasing. I trusted that a Zwave server would do at least the most basic checks which is "does the stick driver load up", clearly I was wrong for giving the project this level of trust and that's on me. But I think users are being reasonable to expect the basics to work, no? Open Source or Free, or not, it doesn't matter, if this is in the hands of the public there should be some responsibility toward those users being safe from critical bugs like this one that could disable safety features. Once trust is broken it's only a matter of time until users abandone a product and the project dies off, so it's in everyone's interest to play nice.

Having said the above I want to express my gratitude for each and every person writing code for this project, you are truly appreciated, I mean that! If all of you writing code could please do basic checks in the future to protect at least those of us who are more vulnerable then you'd be doing a really nice thing indeed.

Thanks.

robertsLando, 2024-11-19:
@Majestic7979 We completely understand that and we are sorry for any inconvenience, we do our best every day to keep everything working as expected, sometimes there could be some unexpected issues but, in most cases, we fix them in a few hours especially if they are bad 🙏🏼

AlCalzone, 2024-11-19:
> I implore the developers to please test with the most common sticks before releasing

Just to be clear, this was not an issue with stick incompatibility. All sticks speak the same protocol, and there are actually hundreds of tests that use this, which all passed.

The main issue was that I switched the way binary data is parsed to a more modern implementation. Strangely, this more modern implementation is missing support for the very basic ASCII encoding, unless the Node.js runtime on that system was compiled with full internationalization support. To be honest, the possibility of this didn't even cross my mind.

And of course, all development machines, as well as the automated testing on GitHub were using runtimes with full internationalization support.

The second issue had to do with how this library is bundled downstream. Even though it is tested here that the released packages are valid, bundling them in that specific way broke at runtime, but only when using Docker, not when running Z-Wave JS UI directly (which I also did extensively).

That said, I'll talk to @robertsLando to see how we can do at least a basic sanity check of the Docker images on CI before releasing them.

On a more personal note: You must be very trusting in technology to install such a lock. For me, everything basic like lights, locks, etc. MUST have a physical fallback to control it. I wouldn't dare to install such a lock.
But I've also had to break a window before (not at home) because a battery-powered non-connected lock wouldn't open anymore.

Majestic7979, 2024-11-22:
> > I implore the developers to please test with the most common sticks before releasing
> 
> Just to be clear, this was not an issue with stick incompatibility. All sticks speak the same protocol, and there are actually hundreds of tests that use this, which all passed.
> 
> The main issue was that I switched the way binary data is parsed to a more modern implementation. Strangely, this more modern implementation is missing support for the very basic ASCII encoding, unless the Node.js runtime on that system was compiled with full internationalization support. To be honest, the possibility of this didn't even cross my mind.
> 
> And of course, all development machines, as well as the automated testing on GitHub were using runtimes with full internationalization support. So this issue was never noticed until it happened in production.
> 
> The second issue had to do with how this library is bundled downstream. Even though it is tested here that the released packages are valid, bundling them in that specific way broke at runtime, but only when using Docker, not when running Z-Wave JS UI directly (which I also did extensively).
> 
> That said, I'll talk to @robertsLando to see how we can do at least a basic sanity check of the Docker images on CI before releasing them.
> 
> On a more personal note: You must be very trusting in technology to install such a lock. For me, everything basic like lights, locks, etc. MUST have a physical fallback to control it. I wouldn't dare to install such a lock. But I've also had to break a window before (not at home) because a battery-powered non-connected lock wouldn't open anymore.

Thank you very much for your polite and detailed reply,as well as to @robertsLando, you guys are great, truly!

Unfortunately it is a tradeoff, I trust Yale/Assa Abloy products, I am actually one of their beta testers (I use this on my back lock which isn't critical). I have used this lock extensively for years without issues, it is indeed very reliable, furthermore mechanical locks can fail, too, so there is always a risk. This is why @robertsLando mentioned that other companies have bugs... I fully agree and did not intend to make anybody feel blamed or guilty, it can happen and happened, it was a user's plight to improve further and check more thoroughly, with a "lessons learned" mindset, and not a critical mindset which only hurts everyone in the process and is unproductive. The fact is I truly depend on your code and it has benefitted me in so many ways greatly, I cannot find words to thank you guys doing this for free and even listening to feedback, I hope it was not received harshly, it was a humble request to ask for more attention during the checks to ensure it is as safe to use as possible. Your product is invaluable to me and I am truly grateful for your time and effort in developing this product. Thank you for fixing it quickly. I have not yet updated, for fear that it might happen again, but at some point I will have to, if only to keep up with requirements, and any issuess I'll report back. Please accept my sincere apology if this has made you feel bad, it was never my intention.

AlCalzone, 2024-11-22:
No offense taken! To be honest, these latest 14.x releases have been more bumpy than I'd like.

Majestic7979, 2024-11-23:
> No offense taken! To be honest, these latest 14.x releases have been more bumpy than I'd like.

I'm glad to know that my message didn't cause any offence. I hope your future releases give you more satisfaction. I'll try to upgrade my server now and see what happens :)

Majestic7979, 2024-11-23:
Pleased to report that tag 9.27.7 loaded just fine, thank you!