zwave-js/zwave-js #7417: Firmware Update fails for Zooz ZEN15 800LR

rcode6, 2024-11-18:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.27.2.b0ade31

### ZwaveJS version

14.3.2

### Describe the bug

The UI shows an available firmware update (2.0.1 to 2.10) for my Zooz ZEN15 800LR.  Clicking the Update button or manually uploading the firmware update results in the same error message: `Error while calling firmwareUpdateOTA: Failed to start the update: Invalid hardware version! (ZW1503)`

### To Reproduce

1. Go to the OTA Updates tab of the device
2. Click UPDATE

or

1. Go to the OTA Updates tab of the device
2. Click ADVANCED
4. Click BEGIN under Firmware update
5. Choose Firmware File, click OK

### Expected behavior

Successful firmware update

### Additional context

Z-Wave device is a Zooz ZEN15 800LR, hardware version 2.0.
Stick is Zooz ZST39 LR on firmware v1.50, sdk v7.22.1

Already contacted Zooz support and confirmed I have the correct hardware version for the firmware update.  They suggested I try filing a bug report here.

AlCalzone, 2024-11-18:
@rcode6 can you please share a diagnostics dump of the device? Click on the device, <kbd>Advanced</kbd>, then export JSON <kbd>Driver</kbd>:
[image]

> Oddly every Z-Wave device I've included since updating to 9.27+ is marked with a re-interview being necessary.

This is expected. I've changed the hash method to detect this. The new code considers the old hash, but the opposite is not possible. Just ignore that until you update again.

rcode6, 2024-11-18:
@AlCalzone here's the dump.  I actually have 4 of these ZEN15s that all exhibit the same issue, but I'm assuming a dump from just one of them is enough for now.
[node_19_dump.json](https://github.com/user-attachments/files/17803465/node_19_dump.json)

pyrodex, 2024-11-19:
I have the exact same issue with a brand new setup.

ZooZ 800LR ZAC93 GPIO off a Raspberry Pi Zero 2
Running 1.50/SDK 7.22.1
Docker install on another node and Zwave-JS talking via TCP socket
zwave-js-ui: 9.27.2.b0ade31
zwave-js: 14.3.2

I just restarted my Z-Wave journey with a ZEN20 800LR Power strip and it currently FW: v4.10.0/SDK: v7.19.3 so I tried to do an OTA update of it fresh out of the box and online and got the same error reported:

`
Failed to start the update: Invalid hardware version! (ZW1503) firmwareUpdateOTA undefined
`

Here is my node dump.

[node_2_dump.json](https://github.com/user-attachments/files/17816104/node_2_dump.json)

davidjmeier, 2024-11-19:
Same issue here with another ZEN15.
-----------------------------------------
zwave-js-ui: 9.27.2.b0ade31
zwave-js: 14.3.2
-----------------------------------------
Docker install external to HomeAssistant.
-----------------------------------------

Attached a node dump.
[node_3_dump.json](https://github.com/user-attachments/files/17819190/node_3_dump.json)

Thank you!

grantland, 2024-11-21:
Tested on latest release and still no luck

zwave-js-ui: 9.27.3.b55a524
zwave-js: 14.3.3
home id: 3778893703
home hex: 0xe13d5787

AlCalzone, 2024-11-21:
I think I have everything to investigate this - just need to find the time. Please no "me too" comments unless you have something new to add.

davidjmeier, 2024-11-24:
Thanks @AlCalzone!  I posted my ZEN15 above, but also see that the issue also impacts ZEN17.  If you'd like the node dump for this let me know, but guessing they're all similar in nature.  Really appreciate all the work on on the project!

AlCalzone, 2024-11-24:
800 series ZEN17 or 700 series? I have the 700 series one, so that would make it easier to reproduce.

rezzalex, 2024-11-26:
Hello,

same issue with a shelly wave shutter : QNSH-001P10

zwave-js-ui: 9.27.7.d3d7049
zwave-js: 14.3.6

this message did not appear a couple of weeks ago, with an older zwavejeUI version, but my node firmware update failed at the time, it took so long that I stopped it...
I precise that the node has been perfectly working since that.

this is the node 80 :

[nodes_dump (1).json](https://github.com/user-attachments/files/17921730/nodes_dump.1.json)

davidjmeier, 2024-11-26:
> 800 series ZEN17 or 700 series? I have the 700 series one, so that would make it easier to reproduce.

It is the 700 series.  It was purchased a couple years ago.  I have a few of them I can do testing on if needed (new in box, unused).

Attached is the node dump.
[node_23_dump.json](https://github.com/user-attachments/files/17925487/node_23_dump.json)

dewoodruff, 2024-11-27:
Also getting this with a ZEN71 800LR on v3.20. Fails with the same error with the suggested update to 3.30, and manually trying to update to 3.40. Dump attached.

zwave-js-ui: 9.27.2.b0ade31
zwave-js: 14.3.2

[node_52_dump.json](https://github.com/user-attachments/files/17927697/node_52_dump.json)

bluewalk, 2024-11-30:
> QNSH-001P10

This, I have the `QNSH-001P10EU` version, but zwave-js is recognizing it as `QNSH-001P10` causing firmware updates to fail. I am reluctant to try flashing the USA version as it will most likely brick the device.

victorgrund, 2024-11-30:
For those in a hurry, it may be helpful to know that I was able to get my ZEN15 firmware updated by temporarily rolling back to zwave-js 13.10.1, zwave-js-ui 13.10.1 (via docker image zwavejs/zwavejsui:9.25.0).   That version was an arbitrary choice on my part, but it worked right away.

mynamewastaken, 2024-12-02:
> For those in a hurry, it may be helpful to know that I was able to get my ZEN15 firmware updated by temporarily rolling back to zwave-js 13.10.1, zwave-js-ui 13.10.1 (via docker image zwavejs/zwavejsui:9.25.0). That version was an arbitrary choice on my part, but it worked right away.

Thanks for this - this worked for me with a Zooz ZAC36 water valve.

AlCalzone, 2024-12-02:
> Seems [devices.zwave-js.io?jumpTo=0x0460:0x0003:0x0082:0.0](https://devices.zwave-js.io/?jumpTo=0x0460:0x0003:0x0082:0.0) is categorizing both under the same product.
> 
> **(0x0082 for the EU version and 0x0083 for the US version)**

Only the actual IDs (and in some cases RF region) are used to look up updates. In your case, this is handled by the firmware update service:
https://github.com/zwave-js/firmware-updates/blob/60a94e51157c3e9926c4258516a8cd117d94a68d/firmwares/shelly/qnsh-001P10.json#L19

bluewalk, 2024-12-02:
> Only the actual IDs (and in some cases RF region) are used to look up updates. In your case, this is handled by the firmware update service: [...]

I see, but unfortunately downloading the EU variant of the firmware myself from the [Shelly website](https://support.shelly.cloud/en/support/solutions/articles/103000258471-shelly-qubino-wave-firmware-updates) and going through `Advanced->Firmware upgrade` throws the error of "Invalid hardware version".

LennyBar79, 2024-12-02:
> For those in a hurry, it may be helpful to know that I was able to get my ZEN15 firmware updated by temporarily rolling back to zwave-js 13.10.1, zwave-js-ui 13.10.1 (via docker image zwavejs/zwavejsui:9.25.0). That version was an arbitrary choice on my part, but it worked right away.

thank you, you saved my time, impossible to update with latest version, and it works with 9.25.0 !

AlCalzone, 2024-12-02:
Confirmed with the Wave Shutter. Fix coming.

closed 2024-12-02 by zwave-js-bot

bluewalk, 2024-12-03:
> Confirmed with the Wave Shutter. Fix coming.

Confirmed, able to update the firmware now. Great work!