zwave-js/zwave-js #7419: Fibaro FGS222 switch state is not getting updated when using the pyhsical switch

nipsgur, 2024-11-18:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.27.2.b0ade31

### ZwaveJS version

14.3.2

### Describe the bug

Since the last update the state of the FGS222 relay switches is not updated, when I use the physical switch.
(I can still switch it with my home automation domoticz via mqtt.)
I noticed it, since the change if the relay switch triggers other lights,
I would be glad about any suggestions,
Torben

### To Reproduce

I press the switch, but in the UI I don't see a change of the binary switch. I have two, it shows up on both,

### Expected behavior

Pressing the switch fo FG222 should change the state of the binary switch in the UI

AlCalzone, 2024-11-21:
This should be fixed in the latest release v14.3.4 (https://github.com/zwave-js/node-zwave-js/pull/7426)
If not, feel free to report back and reopen the issue.

closed 2024-11-21 by AlCalzone

nipsgur, 2024-11-21:
Perfect!
Thanks a lot, both relay switches are reporting their state to the controller after I updated.