zwave-js/zwave-js #7422: Unable to make an NVM backup, the backup freezes

roazhon, 2024-11-20:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [ ] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.20.0

### ZwaveJS version

plugin Jeedom 2024-11-16 01:03:20

### Describe the bug

the backup starts then freezes after a few seconds
Home automation JEEDOM  4.4.19 with VM Debian 12 on NAS QNAP (virtualization station), USB key aeotec Z-Stick Gen5 SDK 6.81 firware 1.2
Everything works well except the possibility of making an NVM backup from the ZwaveJS-UI interface
Please find logs in attachment and a screenshot

### To Reproduce

Other users in jeedom community have related this problem to backup NVM but no solution has been found
They used PC simplicity studio and PC controller but it's not so simple

### Expected behavior

Have a backup NVM is interrested for me to restore on a new key 700 serial

### Additional context

[Archive.zip](https://github.com/user-attachments/files/17828906/Archive.zip)
[image]

robertsLando, 2024-11-21:
@AlCalzone Works on my side

FW: v1.2
SDK: v6.81.6

AlCalzone, 2024-11-21:
So the problem is likely caused by the VM USB setup.
@roazhon do you have the possibility to create the backup on a different computer? E.g. by downloading and running one of the precompiled binaries from https://github.com/zwave-js/zwave-js-ui/releases/tag/v9.27.3 ?

roazhon, 2024-11-22:
hello, thank you for support... I will try to test on a windows pc the windows binary by following this process "How to Run Z Wave JS UI as a Windows Service" https://www.youtube.com/watch?v=6GD-DrpDdXo , could you confirm that this process works ?

AlCalzone, 2024-11-22:
Windows **might** need additional drivers for the stick. Check aeotec's website if you find something.

roazhon, 2024-11-22:
Drivers are installed, my AEOTEC key is detected on device manager on COM3 call UZB
[image]

AlCalzone, 2024-11-22:
Then just enter COM3 as the port.

AlCalzone, 2024-11-22:
Thanks for the update. Since this seems to be caused by something external (probably VM config), I'll close this issue as there's not much we can do about it.

closed 2024-11-22 by AlCalzone