zwave-js/zwave-js #7424: Compat: mapping root reports to endpoint not working

AlCalzone, 2024-11-20:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/7423

<div type='discussions-op-text'>

<sup>Originally posted by **@rohrsh** November 20, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
I have a number of FGS221 (Fibaro Double Relay Switch 2x1.5kW) units. When I press the two physical buttons, only the state of the second channel is updated in ZWaveJS UI, not the first. 

What did you expect to happen instead?
Status reports for first channel to be received and handled by ZWaveJS. 

Steps to reproduce the behaviour: Mash the buttons. 

Anything else we should know?

I presume this is related to the _mapRootReportsToEndpoint_ flag and perhaps can be corrected with a change to the device json. These Fibaro nodes are somewhat old Zwave protocol 3.5 devices, running Fibaro firmware 2.2. 

I tried creating association groups to tie the switches back to the controller, but it did not solve the problem. 

> 

### Software versions

Driver (node-zwave-js): zwave-js: 14.3.2

Z-Wave JS UI: 9.27.2.b0ade31

### Z-Wave Controller (Stick/Dongle/...)

Silicon Labs reference 700 series USB controller - 7..21.5

Note same results on  Zooz 700 FW: v7.19.2 (latest available firmware for AU)

### Device information

Manufacturer: Fibaro
Model name: Double Relay Switch 2x1.5kW | FGS221 FW v2.2
Node ID: 006 (Playroom AC and Fan PUCK)

Manual https://manuals.fibaro.com/content/manuals/en/FGS-221/FGS-221-EN-A-v1.01.pdf 

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [ ] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

In this log file I press button 2 [Endpoint 2] a few times (updates ok), and then button 1 [Endpoint 0] many times (no updates)

[first_channel_silent.log](https://github.com/user-attachments/files/17829567/first_channel_silent.log)

[node_6.json](https://github.com/user-attachments/files/17829612/node_6.json)
[node_6_dump.json](https://github.com/user-attachments/files/17829613/node_6_dump.json)

</div>

rohrsh, 2024-11-20:
Thank you. 

I should have mentioned: when I manually query the node state via REFRESH button in ZWaveJSUI, it does collect the correct state on both channels. 

e.g. See this log file - I press REFRESH two times ... and Endpoint 1 is mentioned/updated.

[zwavejs_2024-11-21.log](https://github.com/user-attachments/files/17836138/zwavejs_2024-11-21.log)

AlCalzone, 2024-11-20:
Update: I can confirm in tests that `node-zwave-js` alone does the right thing. So this might be related to how it's packaged in Z-Wave JS UI.

PR #7426 (2024-11-20): fix: ensure CC classes are not randomly renamed during CJS transform
fixes: #7424

Apparently decorated classes would be renamed internally and have a numeric suffix appended, e.g. `NoOperationCC` -> `NoOperationCC2`, which breaks some of the CC handling code that requires them to end in `...Report`.
closed 2024-11-20 by zwave-js-bot