zwave-js/zwave-js #7456: Failure attempting to update HS-WX300-R2 firmware: The firmware target is invalid

grantland, 2024-12-04:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.27.8.6c82b1d

### ZwaveJS version

14.3.7

### Describe the bug

Failure when updating firmware of HS-WX300-R2:  "The firmware target is invalid"

### To Reproduce

1. Download the most recent firmware for the HS-WX300-R2 from here: https://docs.homeseer.com/products/wx300-firmware-updates
2. Attempt to install it by going to zwave-js-ui > node > advanced > firmware update > begin
3. Select the firmware file and left the target default as "0"

4. See update status 0-100

### Expected behavior

Firmware update successfully

[node_19_dump.json](https://github.com/user-attachments/files/18003579/node_19_dump.json)
[node_19.json](https://github.com/user-attachments/files/18003580/node_19.json)
[zwave-js-ui-store.zip](https://github.com/user-attachments/files/18003590/zwave-js-ui-store.zip)

AlCalzone, 2024-12-04:
According to the Z-Wave specifications, this means that
> The transferred image does not match the Firmware Target [chosen at the start]

Maybe the update is corrupt - try contacting HomeSeer support.

grantland, 2024-12-11:
Turns out this only occurs with one of my HS-WX300-R2 devices and the others were successfully upgraded. I'm now in contact with Homeseer customer support so I'll close this for now.

closed 2024-12-11 by grantland