zwave-js/zwave-js #7458: Consider ignoring thermostat reports for unsupported setpoints

AlCalzone, 2024-12-04:
Related: https://github.com/zwave-js/zwave-js-ui/discussions/4048

friendlycrow, 2025-08-26:
https://github.com/home-assistant/core/issues/150903

AlCalzone, 2025-09-08:
> It not only leads to multiple setpoints on some of my thermostats (they are all the same Popp/Danfoss), but also to other rogue entries, like motion and current sensors. A thermostat only has one setpoint, one temperature sensor and some technical sensors like battery and communication status.

This is typically a symptom of not having encryption enabled together with bad connectivity or high noise levels:
https://zwave-js.github.io/zwave-js/#/troubleshooting/nonsensical-values

We can do something about the thermostat reports, because we can determine the exact set of supported setpoints during the interview.
For sensors this is only possible for newer devices.

In the meantime, I recommend trying to eliminate the underlying issue rather than fighting the symptoms:
https://zwave-js.github.io/zwave-js/#/troubleshooting/first-steps