zwave-js/zwave-js #7464: Missing device configuration: Philips Deadbolt DDL240X-15HZW

frittsy, 2024-12-08:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Philips touchscreen deadbolt

### Manufacturer ID

0x021D

### Product Type

0x0903

### Product ID

0x00D4

### Firmware Version

1.18.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4973

### Do you have a manual?

https://www.documents.philips.com/assets/20231121/6abae07a387b4b81b81fb0c1002cc6de.pdf

frittsy, 2024-12-08:
This is a new manufacturer for the database, as far as I can tell. Not sure if there's anything special to be done for that but just wanted to point it out. I pulled the manufacturer ID from the XML file on ZWA.

hceuterpe, 2024-12-14:
It's not a new manufacturer.  It's a lock made by Kaadas (0x21d), brand licensed under the Philips Home Access line.  Anywho, I am interested in switching to this lock in the near future.

Try out [#7498](https://github.com/zwave-js/node-zwave-js/pull/7498) if you're interested.

AlCalzone, 2025-01-10:
fixed by https://github.com/zwave-js/node-zwave-js/pull/7498