zwave-js/zwave-js #7472: Qubino ZMNHID Flush On/Off Thermostat provides no Min/Max Setpoint Temp Attributes

Nick-Sills, 2024-12-09:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.27.8.6c82b1d 

### ZwaveJS version

14.3.7

### Describe the bug

I'm using a Qubino Flush On/Off Thermostat - ZMNHID - to control the heating in a domestic hot water tank. This is a typical use case for this device. 

https://devices.zwave-js.io/?jumpTo=0x0159:0x0005:0x0051:0.0

ZwaveJS isn't supplying min_temp or max_temp attributes for the climate entity and this is causing Home Assistant to use default climate min = 0 and max = 50 for the climate entities if the values aren't specified by ZwaveJS.  Given that Home Assistant Climate Entities shouldn't really default for Hot Water, this seems pretty reasonable, but the actual min and max for the device is  -25 to 80 degrees Celsius. 

### To Reproduce

Access the climate entity in Home Assistant and try and set a setpoint temperature >50 degrees,  An error is returned, 

### Expected behavior

ZwaveJS to set attributes max_temp to 80 and min_temp to -25 the same as ZMNKID  Qubino / Flush OnOff Thermostat 2

This device works correctly:
https://devices.zwave-js.io/?jumpTo=0x0159:0x0005:0x0054:0.0

### Additional context

The thermostat can be set to 'Cooling mode', hence min = -25 degrees Celsius, and in heating mode a common use case is Hot Water heating, hence max = 80 degrees Celsius. 

This follows exactly how the later version of the thermostat - ZMNKID  behaves

AlCalzone, 2024-12-09:
The problem here is that the device does not support reporting its range, so this info isn't known to Z-Wave JS or HA.
The default range HA uses for this case was previously widened in https://github.com/home-assistant/core/pull/128543 but apparently that's not enough either.

Related: https://github.com/home-assistant/architecture/discussions/1152

Nick-Sills, 2024-12-09:
> The problem here is that the device does not support reporting its range, so this info isn't known to Z-Wave JS or HA. The default range HA uses for this case was previously widened in [home-assistant/core#128543](https://github.com/home-assistant/core/pull/128543) but apparently that's not enough either.
> 
> Related: [home-assistant/architecture#1152](https://github.com/home-assistant/architecture/discussions/1152)

Thanks for the reply. 

That's what I thought. I could be wrong, but it doesn't look like ZMNKID reports its range either. 
https://devices.zwave-js.io/?jumpTo=0x0159:0x0005:0x0054:0.0

however that device is providing Min and Max set temp values. Are they hard coded?  If they are, then the same would fix ZMNHID.

AlCalzone, 2024-12-09:
> I could be wrong, but it doesn't look like ZMNKID reports its range either

It does. It supports Thermostat Setpoint CC V3, which adds the functionality to advertise the range, while the ZMNHID only supports V2 of that CC.

Nick-Sills, 2024-12-09:
> > I could be wrong, but it doesn't look like ZMNKID reports its range either
> 
> It does. It supports Thermostat Setpoint CC V3, which adds the functionality to advertise the range, while the ZMNHID only supports V2 of that CC.

Ahh, OK thanks for the info. I'll have to open a Home Assistant issue then. This is going to be a tough sell because the max default has already been bumped to 50 Celsius and 80 is likely to be seen as out of range for a climate entity default. 

Thanks for your help. It's appreciated.

Nick-Sills, 2024-12-09:
I see you've already linked this issue to - https://github.com/home-assistant/architecture/discussions/1152

Thank you