zwave-js/zwave-js #7482: Remote Z-Wave Driver not working/supported

Oekel, 2024-12-12:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.27.5

### ZwaveJS version

14.3.4

### Describe the bug

The Driver can not be loaded on all Docker Versions 9.26.0 until **newest**.
The Last working one (I rollback to) is 9.25.0 with 13.10.1

I will attack all ScreenShots. Please provide me with additional Informations if there are special taskt to switch from 9.25.0 to a higher version.

`
zwave-js-ui:
    container_name: zwave-js-ui-haberloh
    image: zwavejs/zwave-js-ui:9.25.0
    restart: unless-stopped
    tty: true
    stop_signal: SIGINT
    environment:
      - SESSION_SECRET=mysupersecretkey
      - ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db
      # Uncomment if you want logs time and dates to match your timezone instead of UTC
      # Available at https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
      - TZ=Europe/Berlin
    networks:
      vlan50:
        ipv4_address: 192.168.5.213
    #devices:
      # Do not use /dev/ttyUSBX serial devices, as those mappings can change over time.
      # Instead, use the /dev/serial/by-id/X serial device for your Z-Wave stick.
      #- 'tcp://192.168.1.102:4000'
    volumes:
      - zwave-config:/usr/src/app/store
 `
[image]
[image]
[image]
[image]
[image]
[image]

### To Reproduce

Just switch Version in docker compose

### Expected behavior

As usual just working with no config changes.

AlCalzone, 2024-12-12:
That's not a driver log.

Oekel, 2024-12-12:
> That's not a driver log.
I followed the animation in the link. Then please tell me how I can provide the data.

AlCalzone, 2024-12-12:
[image]

Oekel, 2025-02-16:
[zwavejs_2025-02-16.log.zip](https://github.com/user-attachments/files/18814614/zwavejs_2025-02-16.log.zip)

Is that one the right?

Totay upgradet from 9.25.0 (working)
To 9.30.1 (no ACK to driver)

Hope you can see something in the logs.
Changing timeouts to max did not help

AlCalzone, 2025-02-17:
Looks like a duplicate of https://github.com/zwave-js/node-zwave-js/discussions/7504#discussioncomment-11622515

closed 2025-02-17 by AlCalzone