zwave-js/zwave-js #7493: Missing device configuration: Yale Assure Lock 2 Keyed Deadbolt

harmfulmushroom, 2024-12-13:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Yale / Assure 2 Biometric Deadbolt

### Manufacturer ID

0x0129 

### Product Type

0x8107

### Product ID

0x09d2 

### Firmware Version

1.4.12

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4929/features

### Do you have a manual?

https://support.remotelock.com/s/article/Yale-Assure-Lock-2-Keyed-Deadbolt-Installation-Guide

harmfulmushroom, 2024-12-13:
If it would be helpful I can also provide a copy of the Debug Info, I'll just redact door codes prior.

hceuterpe, 2024-12-14:
This is a curious one...
Are you able to confirm that device id?  Those Yale Biometric locks are implemented here:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0129/yrd4x0-f-zw3.json

The device id of 0x49d2 matches up with the Z-Wave alliance product information.

harmfulmushroom, 2025-05-04:
Is there any additional information I can provide for this issue to try and get some native support for this device in ZWaveJS? The features still work to an extent, but support is hit and miss. I would rather keep this lock on Zwave rather than purchase a zigbee module with better native support.

harmfulmushroom, 2025-09-15:
Any updates on this? Would still be happy to provide any necessary information, I just need a little direction.

bodyscape, 2025-12-31:
Hi, i have this device all the config appear to be working it's just the identification on zwavejs. You can use https://devices.zwave-js.io/?jumpTo=0x0129:0x8104:0x45d2:0.0. Same thing.

The device is the YRD420 model touch 2, but with the zwave module v2 not v3, at least in my case.

[image]

[image]

Let me know if you need anything else.