zwave-js/zwave-js #7502: Another case of Fibaro Dimmer 2 (FGD212) can't be controlled from ZWAVE-JS or ZWAVE-JS-UI

oztourer, 2024-12-17:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.27.8

### ZwaveJS version

14.3.7

### Describe the bug

I have a Mios Vera Edge and have been running the Vera integration in Home Assistant for some years. I want to retire the Vera functionality but use it as a network attached zwave controller. I have successfully stopped the Vera applications and set up ser2net so that I can access the zwave network from zwave-js-ui. That is mostly working well, using the Vera's S0 key in zwave-js-ui so that I do not have to rebuild my network. Most devices work as expected but I have two Fibaro FGD212 in-wall dimmers and these devices are not successfully interviewed. They are correctly recognised as Fibaro Dimmer 2 and their in-built sensors are added to the device that is created in the zwave-js embedded in zwave-js-ui. No dimmer switch entity is created though so I cannot control the dimmer from Home Assistant. The expected configuration entries for the device are also not present in zwave-js-ui.

I have tried re-interviewing the device (debug log attached) and much of the process works as expected but the log does show an error related to "querying Basic CC state":
```
  CNTRLR   [Node 009] Interviewing Basic...
  CNTRLR » [Node 009] querying Basic CC state...
  CNTRLR   [Node 009] Timed out while waiting for a response from the node (ZW0201)
  CNTRLR   [Node 009] No response to Basic Get command, assuming Basic CC is unsupported.
```
zwave-js-ui lists only the following command classes for the interviewed device:
- Multilevel Sensor v4
- Meter v3
- Notification v5
- Manufacturer Specific v2
- Version v2
- Configuration

So no Multilevel Switch and therefore no switch entity available for Home Assistant.

### To Reproduce

Try to include a Fibaro Dimmer 2 (FDG212) in the zwave network and later try to re-interview it. The same error will result in both cases.

### Expected behavior

A new device and list of entities should be added to the Zwave integration and one of the entities should be a switch.

### Additional context

I can nuke or unNuke the Vera at will. All devices work correctly in the Vera UI as well as in Home Assistant using the Vera integration. Switching to ser2net for a network attached zwave controller and using Zwave-JS-UI always gives the same result - most devices work fine with the exception of the Fibaro Dimmer 2.

[debug_interview2.log](https://github.com/user-attachments/files/18159919/debug_interview2.log)
[node_9.json](https://github.com/user-attachments/files/18159926/node_9.json)
[node_9_dump.json](https://github.com/user-attachments/files/18159929/node_9_dump.json)

oztourer, 2024-12-17:
Further information: I have a spare Fibaro Dimmer 2 so I tried including that through zwave-js-ui. It claimed to fail but did in fact add the device. The result was the same as when interviewing existing nodes - sensors recognised but no switch added and no configuration entries under Config. During inclusion I get a number of log messages "Dropping message with invalid payload" so it may be that the Vera embedded zwave controller uses a non-standard protocol, or a protocol not currently recognised by node-zwave-js.

oztourer, 2025-01-02:
I decided to exclude and then include the Fibaro Dimmer 2, which causes it to reset itself to default values. The device now configures itself correctly in zwave-js and I get the expected switches. I do not have any load on one of the switch inputs and had configured that one to activate scenes -maybe this caused the device to be handled incorrectly during inclusion? Note that I should still have switches configured in this case as I use a push-button switch to trigger a scene.

Although I still see the report above as describing a bug the issue is in my case resolved.

AlCalzone, 2025-01-13:
> The result was the same as when interviewing existing nodes - sensors recognised but no switch added and no configuration entries under Config

I've heard this before with Fibaro devices. It might have been included with encryption before and was not properly removed or it did not reset its state internally. As a result, controllable CCs are omitted:
```
supported CCs:
· Z-Wave Plus Info
· Version
· Manufacturer Specific
· Association Group Information
· Powerlevel
· Application Status
· Multilevel Sensor
· Meter
· Notification
· CRC-16 Encapsulation
· Security
· Firmware Update Meta Data
```

> During inclusion I get a number of log messages "Dropping message with invalid payload"

Could also be due to bad connectivity corrupting frames. Your logs show that the communication is falling back all the way to 9.6 kbps, although the device should support 100 kbps. You may want to give https://zwave-js.github.io/node-zwave-js/#/troubleshooting/first-steps a look.

oztourer, 2025-01-13:
Security might be relevant. When I researched converting the Vera Edge to a network-attached controller I found information on retrieving the S0 key but not any other keys. So my Zwave JS UI configuration has the original S0 key but freshly generated S2 keys.

### "Dropping message with invalid payload"

I tried many times to re-interview the node and received this message every time.

The dimmer has a direct neighbour association with the controller and is in the same room so I would not expect communication to be slow. I do not have any particularly chatty devices on the network. The whole network is actually very responsive since I switched from the Vera as controller to Zwave JS UI.

AlCalzone, 2025-01-13:
If it still happens, feel free to share another driver log which contains this message, so I can take a look.

oztourer, 2025-01-13:
Thanks, but I have no more issues since resetting the dimmer, deleting the old ZUI device instance and including it once more. The reset will have cleared my configuration on the device, which was non-standard as I was using one of the switches to trigger scenes rather than switch a load, so it is unclear whether the issue was security or this uncommon configuration.

As far as I recall the dimmer did not use the security features in the Vera Edge setup (and still does not on ZUI) so that may or may not be the issue. No way to answer that since the device has been reset. 

Interestingly the dimmer works fine to trigger scenes instead of switch loads on ZUI, with no config changes required, whereas I remember reading originally that the Fibaro dimmer needed a config change to handle this reliably. It has been rock solid since the move away from Vera Edge.

Given that there is no conclusive way at this point to confirm a bug in the ZUI code I am happy for this issue to be closed if you wish.

closed 2025-01-13 by AlCalzone