zwave-js/zwave-js #7506: Missing device configuration: MCOHome MH4936

jprates, 2024-12-19:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

MCOHome MH-4936

### Manufacturer ID

0x015f

### Product Type

0x8101

### Product ID

0x4936

### Firmware Version

2.0.8

### Do you have a manual?

[MH4936.pdf](https://github.com/user-attachments/files/18195882/MH4936.pdf)

jprates, 2025-02-06:
Is there anyway I can help with this issue?

Still not implemented on:

Core 2025.2.0
Supervisor 2025.02.0
Operating System 14.2
Frontend 20250205.0
Z-Wave JS UI 3.20.0

closed 2025-05-12 by AlCalzone

jprates, 2025-05-12:
The MH4936 is a thermostat that can get its power from both 4 x AA batteries or from a 24V AC line.

A couple of days ago I asked MCOHome if they could change the firmware for this thermostat so as to would allow it to act as a repeater when fed by 24V AC.

As it was on firmware v2.0.8 it appeared as solely a battery powered device obviously with isListening set to false.

Coincidentally to this commit I just got today a firmware update (v2.1.8) that supposedly allows for this, I need to check if they implemented things right, didn't have time to check it yet.

Just wanted to give @philh30 and @AlCalzone this heads up and ask if this new feature needs adaptation at your implementation, and if I can help you guys in any way doing so. If no change is needed, then even better.

A big thanks to you both for taking the time to implement this device.

AlCalzone, 2025-05-12:
No, that stuff is handled by the Z-Wave protocol/firmware. You will have to exclude and re-include it though, because this changes the network role of the device.

AlCalzone, 2025-06-02:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of a re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

philh30, 2025-06-02:
The product id for node 6 (`0x4a36`) doesn't match what you posted when you created this issue (`0x4936`). Do you have multiple of these thermostats with different fingerprints?

```
2025-06-02 11:31:03.492 DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                 │ RSSI: -78 dBm
                                 └─[Security2CCMessageEncapsulation]
                                   │ sequence number: 69
                                   │ security class:  S2_Unauthenticated
                                   └─[ManufacturerSpecificCCReport]
                                       manufacturer id: 0x015f
                                       product type:    0x8101
                                       product id:      0x4a36
```

jprates, 2025-06-02:
Interesting, and thanks @philh30 for the feedback.

I need to check with MCOHome support, but I'm guessing they changed the id on the new firmware they sent me when asked if they could do a new version with repeater capability (to use only when fed by AC24V).

Probably decided to assign a virtual id to make no confusion with the original one, again I'm just guessing.
I'll ask them, since I just checked and all 12 thermostats are with id 18998 now as you mentioned for node 6.

jprates, 2025-06-03:
It was a good guess, here's the answer from MCOHome:

"_Why we indicate 4a36 is to distinguish the standard version (MH-4936, which supports battery powered and 24VAC powered)._"

I would therefore suggest that instead of replacing 0x4936 by 0x4a36 we keep both, since the device is basically the same, all that differs is the firmware that uses different repeater attributes according to power source, batteries or AC line.

What do you guys think?

PR #7765 (2025-05-12): feat(config): add McoHome thermostats MH4936, MH5-2D and MH5-4A
Fixes #7506
Fixes #7281
Fixes #6587

Device capabilities based on user manuals attached to those issues and those found [here](https://mcohome.com/DownList/1.html). There are several other similar thermostats at that link but I can't find the fingerprints for them.

Templates for parameters added based on those in the database for the MH5-2A. With the templates in place, the config for the MH5-2A will change slightly as the user manual shows Celsius as the default. The MH4936 is the only one of these four thermostats that claims t [... cut]