zwave-js/zwave-js #7516: ZAC93 Inclusion Stops at status = NodeFound

DaAwesomeP, 2024-12-23:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

14.3.7

### ZwaveJS version

9.29.0.b8373a3

### Describe the bug

When including devices on the ZAC93, the device is found but just stops at `handling add node request (status = NodeFound)` without interviewing the device. I have a similar issue with Smart Start (says NWI matches but doesn't do anything after that). Smart Start LR fails entirely (claims NWI does not match and ignores it).

See log below.

### To Reproduce

1) Update the ZAC93 to FW v1.5.0 (SDK v7.22.1)
2) Power cycle the ZAC93 once complete to ensure it reboots
3) Restart ZWaveJS UI
4) Attempt to include a device

### Expected behavior

Device should be interviewed

### Additional context

[a_zwavejs_2024-12-21.log](https://github.com/user-attachments/files/18217376/a_zwavejs_2024-12-21.log)

AlCalzone, 2025-01-10:
I can only suspect there's something really wrong with the controller firmware. After NodeFound, it should be communicating with the node to actually include it, but that does not seem to happen.

I've also never seen -128 dBm as an RSSI reading:
```
2024-12-21T05:12:37.015Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -128 dBm
                                    channel 1: -128 dBm
                                    channel 2: -128 dBm
                                    channel 3: -128 dBm
```
typically, 800 series controllers are around -110.

hceuterpe, 2025-01-14:
What case did the GPIO module go into? 

Otherwise I'm guessing maybe the controller's region setting might be set wrong for your locale? I could see that being indicative as to why LR failing entirely and say, it's set to a region without LR...

DaAwesomeP, 2025-01-14:
It's on a PoE ESP32 right now (serial over TCP), but I have a USB UART adapter on the way so I can remove that variable. Unfortunately I probably won't be able to try it out until the end of the month though.

This is actually my second ZAC93 (warranty replacement). The first one's radio ceased working entirely after I tried enabling LR. It was working OK (but not great) until I did that. I could never get Smart Start to work and adding devices was finicky. I set region to US w/ LR explicitly and then it stopped working (responded to serial fine but could not activate or deactivate anything). I could change to non-US regions, but I could not reset the region to US w/o LR.

Since this is the second device, I am definitely suspecting an issue with them unless there is egregious corruption/packet loss with the ESP32 (seems unlikely but I will verify).

AlCalzone, 2025-01-14:
> corruption/packet loss with the ESP32 (seems unlikely but I will verify).

There has been at least one issue before where this was the case and caused all sorts of problems. You may be able to eliminate the possibility by using a USB to UART converter and connecting to the host machine directly for a test.

ShawnCollinge, 2025-04-01:
So I plugged my ZAC93 into a raspberry pi and it works fine there so I'm thinking it's this tubesZB esp32 that is having an issue.

hceuterpe, 2025-04-13:
> I have the same setup, Zooz ZAC93 in a TubesZB serial-to-ip adapter and it has been working fine for months, this past Monday however it went belly up and I can't figure out why? I have since reset almost all my devices and tried to add them to a new network setup on a Zooz ZST39 and I'm seeing similar issues, timeouts, etc. including all channels at -128 dBm.
> 
> I've manually entered all the DSK that I have for my devices so SmartStart would work and I'm seeing them in the debug logs appear, "Including Smart Start node with DSK ......" but then it never gets added. I've try adding some zwave plus nodes without SmartStart manually and Inclusion never finds them even thought they are 10 feet from the controller.
> 
> HA Core: 2025.4.2 Supervisor: 2025.04.0 Operating System: 15.1 Z-wave JS UI Add-on: 4.0.4 Zooz ZST39 LR: FW v1.50 SDK: v7.22.1
> 
> Soft Reset: Disabled Controller Recovery: Enabled Watchdog: Enabled Increase node report timeout: Enabled
> 
> [debug.log](https://github.com/user-attachments/files/19726839/debug.log)

Maybe something to consider:  Apparently that adapter uses the Olimex ESP32 board without galvanic isolation and it's actually confirmed in their Q&A. Found at least one [link](https://www.rtautomation.com/industrial-library/serial-device-isolation-grounding) that mentions potential serial port communications issues arrising without it.  Olimex has a [board](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) with that.  I wonder if that could solve this issue?

hceuterpe, 2025-04-13:
> > Maybe something to consider: Apparently that adapter uses the Olimex ESP32 board without galvanic isolation and it's actually confirmed in their Q&A. Found at least one [link](https://www.rtautomation.com/industrial-library/serial-device-isolation-grounding) that mentions potential serial port communications issues arrising without it. Olimex has a [board](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) with that. I wonder if that could solve this issue?
> 
> That could have been part of the issue yeah, I pulled the ZAC93 from the esp32-poe setup and put in a rPI and it gets back to -90 dBm AND I could communicate with my network again. Well mostly, I've reset devices/excluded them/etc. so now I have to do around and repair some.
> 
> Here is the weird part though once smartstart had "stablized" and old nodes were added back I shut the PI down and moved the zac93 back to the esp32-poe board. Started the pi back up and pointed the z-wave-js config at that tcp port for the controller and it's working.

A raspberry pi would be powered via 5V power source, the ZAC93 taking 3.3V.  I mean that's not much of a difference. On the otherhand, for PoE that's upwards of 48V.  I wonder if that's why it works in the Pi, but not this adapter?

DaAwesomeP, 2025-04-13:
Finally able to follow up. I tried with a direct UART connection on a separate machine, and I am still having issues with the ZAC93.

> I have the same setup, Zooz ZAC93 in a TubesZB serial-to-ip adapter and it has been working fine for months, this past Monday however it went belly up and I can't figure out why? I have since reset [...]

@bmorgenthaler this is very similar to the behavior I am seeing. I have _never_ gotten SmartStart to work with the ZAC93, even after they replaced my previous one.

> Maybe something to consider: Apparently that adapter uses the Olimex ESP32 board without galvanic isolation and it's actually confirmed in their Q&A. Found at least one [...]

@hceuterpe I'm actually using the galvanically isolated PoE one. I made my own enclosure for it since it is a bit longer. I have tried both PoE and USB power for the ESP, but now after failing with a direct USB UART into my computer I am somewhat certain it's not the ESP causing the issue (maybe it is making it worse but it isn't the only issue).

> A raspberry pi would be powered via 5V power source, the ZAC93 taking 3.3V. I mean that's not much of a difference. On the otherhand, for PoE that's upwards of 48V. I wonder if that's why it works in the Pi, but not this adapter?

There are 3.3V regulators on the board. The ESP32 runs with 3.3V GPIO and the ZAC93 is powered from 3.3V.

> Also considering the ZAC93 would also have a radio, wouldn't be surprised if it would be even more impacted by this?

I'm more wondering if there is some electromagnetic noise from the Olimex board interfering with the radio on the ZAC93. I have the radio on my ESP32 entirely off (no WiFi or Bluetooth), but I am not sure if the Olimex board is FCC (or other country equivalent) certified.

hceuterpe, 2025-04-14:
> Finally able to follow up. I tried with a direct UART connection on a separate machine, and I am still having issues with the ZAC93.
>
>
> [@bmorgenthaler](https://github.com/bmorgenthaler) this is very [...]

Yeah no on the FCC. Their datasheet even has this [link](https://www.olimex.com/Products/Disclamer/evaluation-board-notice/).  This seems kinda sleazy, but they're basically saying you shouldn't even be using it, nor should it have been sold to you in a kit.  This is probably a lost cause...

kavo91, 2025-05-22:
> > I'm having this issue as well with a TubesZB POE device
> 
> What radio are you using? 

I'm using the ZAC93 on the TubesZB Zwave poe kit. First Zwave device. I emailled Tubes and got this: "Thanks for reaching out. I haven't encountered the "NodeFound" issue with the ZAC93 very often. In the few instances where it did happen, replacing the ZAC93 seemed to resolve the problem."