zwave-js/zwave-js #7520: Missing device configuration: Z-Wave.Me Z-Station

Fohdeesha, 2024-12-24:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Z-Wave.Me Z-Station

### Manufacturer ID

0x0147

### Product Type

0x0401

### Product ID

0x0301

### Firmware Version

7.40

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/5038?selectedFrequencyId=2

### Do you have a manual?

[Z-Station Manual web.pdf](https://github.com/user-attachments/files/18240228/Z-Station.Manual.web.pdf)

Fohdeesha, 2024-12-24:
went ahead and submitted a PR: https://github.com/zwave-js/node-zwave-js/pull/7521

Botched1, 2025-01-09:
I still say if @poltos would release a version that used STANDARD SiLabs zwave firmware instead of his proprietary firmware, so that normal NVM backups and firmware update processes could be used, he would sell 20x more of these.

The form factor is GREAT, but only having "kind of, sort of" support of it in zwave-js (again backups and firmware updates) is a deal breaker.

Fohdeesha, 2025-01-09:
I mean, you're not wrong. That said I'm able to perform an NVM backup in zwave-js, at least it seems to work and I get a file. Am I missing something? 

So, given its silabs silicon, is flashing standard firmware an option? I've developed quite a lot of crossflashing/bootloader replacement/etc guides out there, but all for network equipment mostly. Haven't poked around at zwave hw (yet). I see bootloader stuff seems to be documented at least

Botched1, 2025-01-09:
I was explicitly told that NVM backups done in zwave-js likely would not restore correctly to his custom firmware, but maybe there was a language barrier and I misunderstood or maybe the situation has changed (it was closer to when the z-station came out and I received mine that I asked the question).

I have thought that flashing the standard bootloader and firmware may be an option (likely even), but I never tried it and of course zwave.me wouldn't comment either way when asked as they don't want to support/be liable for it if it gets bricked.

closed 2025-01-10 by AlCalzone