zwave-js/zwave-js #7534: Fibaro FGFS101 Flood Sensor only includes only Node Status and Binary

RonnieVa, 2025-01-07:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

7.22.2

### ZwaveJS version

7.22.2

### Describe the bug

When I include the Fibaro Flood sensor FGFS-101 into my Homeseer Z-wave series 800 controller, only the Node status and Binary are included. The Battery, Temp and tamper are not included. Also when the Flood sensor goes into alarm mode, the information is not passed to Homeseer. I have a Fortrezz water Valve and there is not an option to include the Fortrezz water valve in the Associations.

Thank you,

Ron

### To Reproduce

Include a Fibaro Flood sensor into a Zwave-JS network

### Expected behavior

When the sensor is violated, Homeseer should toggle to alarm mode and I should be able to link my Fortrezz Water Valve in Associations.

didros, 2025-01-12:
Exact same issue here, with version 25.25, which was working fine with openzwave. I moved to zwavejs and since no way to have temperature working.
Btw, I can't see that the commandClass of the temparature in the dump from @nsd0dkassen. It should be 49.

AlCalzone, 2025-01-12:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of a re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

noXi0815, 2025-01-14:
hi, will join the party. have also issues with version 25.25. is this the log you ask for?

`2025-01-14T19:00:14.423Z CNTRLR   [Node 019] Re-interview scheduled, waiting for node to wake up...
2025-01-14T19:00:42.042Z DRIVER   one or more queues busy
2025-01-14T19:00:42.043Z SERIAL » 0x0103003bc7                                                         (5 bytes)
2025-01-14T19:00:42.043Z DRIVER » [REQ] [GetBackgroundRSSI]
2025-01-14T19:00:42.047Z SERIAL « [ACK]                                                                   (0x06)
2025-01-14T19:00:42.049Z SERIAL « 0x0107013b9f9797a0fd                                                 (9 bytes)
2025-01-14T19:00:42.049Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:00:42.050Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -97 dBm
                                    channel 1: -105 dBm
                                    channel 2: -105 dBm
                                    channel 3: -96 dBm
2025-01-14T19:00:42.051Z DRIVER   all queues idle
2025-01-14T19:01:05.604Z SERIAL « 0x011400498400130d04a102309c60858e7270868084d8                      (22 bytes)
2025-01-14T19:01:05.605Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:01:05.606Z DRIVER « [Node 019] [REQ] [ApplicationUpdateRequest]
2025-01-14T19:01:05.606Z CNTRLR « [Node 019] Received updated node info
2025-01-14T19:01:05.607Z CNTRLR   [Node 019] The node is now awake.
2025-01-14T19:01:05.608Z CNTRLR   [Node 019] Node does not send unsolicited updates; refreshing actuator and sen
                                  sor values...
2025-01-14T19:01:05.611Z CNTRLR   [Node 019] [+] [Node Naming and Location] name: "Wasser Sensor Küche"
2025-01-14T19:01:05.612Z CNTRLR   [Node 019] [+] [Node Naming and Location] location: "Küche"
2025-01-14T19:01:05.613Z CNTRLR   [Node 019] Beginning interview - last completed stage: None
2025-01-14T19:01:05.614Z CNTRLR   [Node 019] new node, doing a full interview...
2025-01-14T19:01:05.614Z CNTRLR » [Node 019] querying protocol info...
2025-01-14T19:01:05.616Z DRIVER   one or more queues busy
2025-01-14T19:01:05.617Z SERIAL » 0x010500410013a8                                                     (7 bytes)
2025-01-14T19:01:05.617Z DRIVER » [REQ] [GetNodeProtocolInfo]
                                    payload: 0x0013
2025-01-14T19:01:05.622Z SERIAL « [ACK]                                                                   (0x06)
2025-01-14T19:01:05.624Z SERIAL « 0x010a0141539c0004a10200dd                                          (12 bytes)
2025-01-14T19:01:05.624Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:01:05.625Z DRIVER « [RES] [GetNodeProtocolInfo]
2025-01-14T19:01:05.627Z CNTRLR « [Node 019] received response for protocol info:
                                  basic device class:    Routing End Node
                                  generic device class:  Alarm Sensor
                                  specific device class: Routing Alarm Sensor
                                  node type:             End Node
                                  is always listening:   false
                                  is frequent listening: false
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     40000 kbps
                                  protocol version:      3
2025-01-14T19:01:05.628Z CNTRLR   [Node 019] Interview stage completed: ProtocolInfo
2025-01-14T19:01:05.628Z CNTRLR » [Node 019] querying node info...
2025-01-14T19:01:05.629Z SERIAL » 0x01050060001389                                                     (7 bytes)
2025-01-14T19:01:05.630Z DRIVER » [Node 019] [REQ] [RequestNodeInfo]
                                    node id: 19
2025-01-14T19:01:05.634Z SERIAL « [ACK]                                                                   (0x06)
2025-01-14T19:01:05.636Z SERIAL « 0x01040160019b                                                       (6 bytes)
2025-01-14T19:01:05.637Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:01:05.638Z DRIVER « [RES] [RequestNodeInfo]
                                    was sent: true
2025-01-14T19:01:06.487Z SERIAL « 0x011400498400130d04a102309c60858e7270868084d8                      (22 bytes)
2025-01-14T19:01:06.487Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:01:06.488Z DRIVER « [Node 019] [REQ] [ApplicationUpdateRequest]
2025-01-14T19:01:06.489Z CNTRLR « [Node 019] node info received
                                  supported CCs:
                                  · Binary Sensor
                                  · Alarm Sensor
                                  · Multi Channel
                                  · Association
                                  · Multi Channel Association
                                  · Manufacturer Specific
                                  · Configuration
                                  · Version
[... 2017 lines cut ...]
2025-01-14T19:01:35.317Z SERIAL » 0x0103003bc7                                                         (5 bytes)
2025-01-14T19:01:35.318Z DRIVER » [REQ] [GetBackgroundRSSI]
[... cut ...]
2025-01-14T19:02:05.323Z SERIAL » [ACK]                                                                   (0x06)
2025-01-14T19:02:05.324Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -99 dBm
                                    channel 1: -104 dBm
                                    channel 2: -104 dBm
                                    channel 3: -99 dBm
2025-01-14T19:02:05.324Z DRIVER   all queues idle
2025-01-14T19:02:14.849Z DRIVER   destroying driver instance...
2025-01-14T19:02:14.872Z DRIVER   driver instance destroyed`

closed 2025-01-15 by zwave-js-bot

pstanar, 2025-01-15:
Would it be possible to include firmware 26.26, which has same problem?

AlCalzone, 2025-01-15:
Sure - I wasn't aware of this before. Just to be sure it's the same underlying issue, can you make a driver log of a re-interview?

pstanar, 2025-01-15:
It is same issue, but I will provide driver log this weekend when I return home.

pstanar, 2025-01-17:
Here is the driver log log for firmware v26.26

[fgfs101.log](https://github.com/user-attachments/files/18458505/fgfs101.log)

AlCalzone, 2025-01-17:
=> https://github.com/zwave-js/node-zwave-js/pull/7561