zwave-js/zwave-js #7536: Zniffer: Add new commands for selecting LR channels

AlCalzone, 2025-01-10:
The latest Zniffer firmware allows selecting LR channels (Classic + A, Classic + B, A + B). We should add support for that.

hceuterpe, 2025-01-13:
Any chance this would expose channel 3 background RSSI value in HA?  Currently I noticed that only 0, 1, 2 background RSSI values get polled with my controller.

AlCalzone, 2025-01-13:
[Zniffer](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/zniffer?id=creating-a-zniffer) is unrelated to your issue. Channel 3 is Z-Wave Long Range and only "exists" when your controller is set to an LR capable region.

hceuterpe, 2025-01-14:
Weird.  It's definitely set to US LR since 67 of my 93 devices are currently included via LR. 
 The background RSSI channel 3 value shows up in Z-Wave JS logs, but HA for some reason doesn't even pick it up (just 0-2)...

closed 2025-03-10 by zwave-js-bot