zwave-js/zwave-js #7546: FGD-212 not reporting Scene Activation Command Class

Matsuo3rd, 2025-01-13:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Manually built (git clone - npm run build - npm install

### Z-Wave JS UI version

9.29.0

### ZwaveJS version

14.3.7

### Describe the bug

FGD-212 (both FW 3.4 and 3.5) are not reporting (at beginning) Scene Activation v0 Command Class capabilities.
It will report this capability once the buttons attached to the device are clicked multiple times (triggering an actual scene).
This is problematic for users (as me) using openHAB ZWave-JS binding (under development) leveraging ZWave-JS debug JSON to introspect nodes configuration + command classes.
It looks like FGD-212 has special treatment regarding Scene Command Class which might be related to this issue.
See:
https://github.com/zwave-js/node-zwave-js/pull/1849
https://github.com/zwave-js/node-zwave-js/pull/1848

### To Reproduce

Fresh Z-Wave JS UI instance hooked with Controller + 1 FGD-212 device.

Look at node info: no Scene Activation v0 reported
[image]

Then double/triple click a button attached to the FGD-212 device.
Look at node info: Scene Activation v0 is reported
[image]

### Expected behavior

 Scene Activation v0 capability shall be reported at the beginning so that it is exposed in debug.json which is used by 3rd-party integration to introspect and reflect node capabilities.

Matsuo3rd, 2025-01-19:
See driver log attached.
FGD-212 devide is NODE 008.
Re-Interview process started at 2025-01-19T18:36:02.574Z (line 4377)

[zwavejs_current.log](https://github.com/user-attachments/files/18469974/zwavejs_current.log)

robertsLando, 2025-02-03:
Should be fixed in latest release. Try to upgrade to zui 9.30.0

Matsuo3rd, 2025-02-03:
@robertsLando: thanks for coming back on this.

Unfortunately, I cannot see any improvements under 9.30.1 (see screenshot). The "Scene Activation v0" capability still shows up only after having triggered a scene from FGD-212 device.

Any indication you can provide on why you are thinking that this issue would have been solved with 9.30.0+ version? Thanks.

[image]

AlCalzone, 2025-02-03:
Sorry, that recommendation is on me. I thought we added that fix for the FGD-212, but it was a different fix for a different Fibaro device.