zwave-js/zwave-js #7566: Fortrezz Water Sensor WWA-02 does not include data

RonnieVa, 2025-01-20:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.26.0

### ZwaveJS version

9.26.0

### Describe the bug

I have several Fortrezz WWA-02 water sensors and when I include them into my Z-wave network, Temperature and sensor On/Off does not appear. The Battery does appear but is always 99%. When the sensor is triggered there isn't any status change.

### To Reproduce

Rescan device

### Expected behavior

I should be able to see the Temperature, Battery Status and whether the sensor is in water detection mode or no water detected.

### Additional context

[Fortrezz WWA-02 Water Sensor.txt](https://github.com/user-attachments/files/18465291/Fortrezz.WWA-02.Water.Sensor.txt)

RonnieVa, 2025-01-20:
Hi, Thank you. Here is 2 different log files. I hope this is what is need to solve my issue.

Thank you again.

[Fortrezz log file 1.txt](https://github.com/user-attachments/files/18480479/Fortrezz.log.file.1.txt)

[Fortrezz log file 2.txt](https://github.com/user-attachments/files/18480482/Fortrezz.log.file.2.txt)

RonnieVa, 2025-01-20:
> Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

[Fortrezz log file 1.txt](https://git

[Fortrezz log file 2.txt](https://github.com/user-attachments/files/18480520/Fortrezz.log.file.2.txt)

hub.com/user-attachments/files/18480517/Fortrezz.log.file.1.txt)

HA36823, 2025-02-04:
I have the same problem. I hope the problem can be resolved soon. I worry that I might have a water leak before it is resolved.

robertsLando, 2025-02-04:
@AlCalzone has been very busy in last weeks, he will answer as soon as possible

RonnieVa, 2025-03-10:
Has there been any updates to this problem?

Thank you.

robertsLando, 2025-03-11:
cc @AlCalzone

Alboats77, 2025-03-18:
I also have the Fortrezz water sensors which do not change status when water is detected. I hope this can be fixed soon.

RonnieVa, 2025-03-21:
Hi,
I am having the same problem, sensor detects water and nothing changes in the status within the Zwave network. The temperature used to work but it no longer updates the temperature.

SCHEAT796, 2025-03-21:
Hi,

I have the same problem that when the Fortrezz WWA-02 water sensor detects moisture there isn't any indication within the zwave network that there has been an event. The temperature is also not updating in the network.

Thanks

Alboats77, 2025-03-24:
Has there been any updates on this problem?  I hate to be a pest but I have flood sensors in my house that will not communicate with my shut-off valve if I have a flood.

Thank you.

AlCalzone, 2025-03-25:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
The log should at least include a re-interview of the device.

RonnieVa, 2025-03-25:
Here is the debug file. Please let me know if you need anything else.

Thank you very much.

[Fortrezz Debug file #2.txt](https://github.com/user-attachments/files/19448860/Fortrezz.Debug.file.2.txt)

AlCalzone, 2025-03-25:
Please read what I've posted. That's not a driver log on level `debug`.

RonnieVa, 2025-03-25:
Here is the correct file (I hope) the Z-wave device is # 015 and the re-interview is near the bottom of the file.

Thank you again

[zwavejs_2025-03-25.log](https://github.com/user-attachments/files/19449714/zwavejs_2025-03-25.log)

AlCalzone, 2025-03-26:
I see this information in the logs:
- Air temperature sensor => Multilevel Sensor
- Heat alarm => Alarm Sensor
- Water Leak alarm => Alarm Sensor

Where exactly is it missing? Z-Wave JS UI? Home Assistant? Something else?

I also see that the device is triggered a few times and that this information is refreshed. However the reported status does not change, but we have no influence on that.

RonnieVa, 2025-03-26:
I am using Homeseer 4.2.22.4 and recently upgraded to zwave+ so now when the device is set into alarm, the status does not change in Homeseer. Attached is a screenshot. The highlighted is Node 15 and the devices above that are from before I upgraded to Zwave+. They do not change status when there is an event either.

Thank you for your help.

[image]

AlCalzone, 2025-03-26:
I think this is something you need to ask HomeSeer. It almost sounds like they have no support for the Alarm Sensor CC in that UI. I wonder why the Multilevel Sensor is missing though...

RonnieVa, 2025-03-26:
Al,

Thank you for everything!!! I have contacted Homeseer and provided a link to this thread. I hope that Homeseer can help fixing this issue.

Thanks for all of your hard work.

closed 2025-04-06 by zwave-js-assistant[bot]