zwave-js/zwave-js #7579: Enhance `Keypad unlock operation` notification with userCode

karmingc, 2025-01-29:
**Is your feature request related to a problem? Please describe.**

Currently, whenever a lock is unlocked by a usercode, the `Keypad unlock operation`  notification is created with the `parameters: userId: {value}`. However, we need to look at the users tab to confirm which code it was used. This may work for most case except the master key.

We recently tested with a Yale Assure Touchscreen Deadbolt YRD226 and when the master key was used to unlock it, the `userId` that was provided was actually in a `disabled` in the `scheduling` column in the `users` tab.

**Describe the solution you'd like**

1. The `Keypad unlock operation` enhancing the `parameters` field with `userCode`. 
2. The `Keypad unlock operation` would include a parameter to indicate whether the userId/userCode represents the master key.

**Describe alternatives you've considered**

Open to suggestions!

**Additional context**

karmingc, 2025-01-30:
Hi @AlCalzone,

The opt-in would be good with our team!

I was able to perform the `User Code CC` command `get` on userId `251` and it now shows on the UI.

However, I also tried the command `getAdminCode` and it returned `FailedZWaveCommand: zwave_error: Z-Wave error 302 - Node #43 does not support the command AdminCodeGet! (ZW0302)`.

With the given information, would there still be a way to deterministically know whether a code was a master code with the current device?

AlCalzone, 2025-01-30:
> With the given information, would there still be a way to deterministically know whether a code was a master code with the current device?

With this exact device yes, by checking if the user ID is 251. Generically for all User Code CC V1 locks, no.