zwave-js/zwave-js #7595: ser2net not working works after like 10 to 20 minutes sometimes

blair287, 2025-02-03:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.29.1

### ZwaveJS version

14.3.7

### Describe the bug

I am using ser2net to host my zwave stick im also doing the same for zigbee.

zigbee is working fine HA ZHA connect to it everytime first time no 1 issue.

however zwave js ui will not connect but will sometimes after a long while.

port: con01
  accepter: tcp,20108
  enable state: on
  timeout: 0
  connected to: ipv6,::ffff:192.168.1.31,34080
    bytes read from TCP: 26441
    bytes written to TCP: 59188
  device: serialdev, /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2151441-if00, 115200n81,nobreak,local
  device config: 115200N81,CLOCAL
  device controls: RTSHI,DTRHI
  tcp to device state: waiting output
  device to tcp state: waiting output
  bytes read from device: 58990
  bytes written to device: 26441
port: con02
  accepter: tcp,20208
  enable state: on
  timeout: 0
  device: serialdev, /dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_533D004242-if00, 115200n81,nobreak,local
  device config: 115200N81,CLOCAL
  tcp to device state: waiting input
  device to tcp state: waiting input
  bytes read from device: 0
  bytes written to device: 0

this is ser2net you can see there is a connection made for the zigbee stick but no connection to the zwave stick

[image]

zwave shows driver failed to initialize. Sometimes with random rebooting of zwave js ui and the ser2net pi or just waiting for a long while it will connect and then work ok until the connection is stopped again either by the pi with ser2net rebooting or zwave js ui addon restarting.

### To Reproduce

connect zwave controller via ser2net

### Expected behavior

to connect

AlCalzone, 2025-02-03:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

Please also share your ser2net config.

blair287, 2025-02-07:
> Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
> 
> Please also share your ser2net config.

Will do at work this week, did update to the 3.20 update and it connected first time.

Will do some reboots and u0load the info you asked for just might be a few days before I can get it thanks.

blair287, 2025-03-01:
> Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
> 
> Please also share your ser2net config.

Sorry it took a while to reply here is the log around the 19:50ish mark is it not connecting to ser2net. zigbee is connected fine. Zwave will connect but it takes anywhere from 10 to 2 hours for it to come back.

[zwavejs_2025-03-01.log](https://github.com/user-attachments/files/19040575/zwavejs_2025-03-01.log)

```
connection: &con02
    accepter: tcp,6000
    enable: on
    options:
      banner: *banner
      kickolduser: true
      telnet-brk-on-sync: true
      chardelay-min: 0
      chardelay-max: 0
    connector: serialdev,
              /dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_533D004242-if>
              115200n81,nobreak,local
```

blair287, 2025-03-01:
> Turn off your banner.
> 
> https://zwave-js.github.io/zwave-js/#/usage/tcp-connection?id=configuring-ser2net
> 
> [#7482](https://github.com/zwave-js/zwave-js/issues/7482) [#7504 (reply in thread)](https://github.com/zwave-js/zwave-js/discussions/7504#discussioncomment-11622515)

Thanks that seems to have fixed it!

closed 2025-03-01 by blair287