zwave-js/zwave-js #7655: Id-Lock 150 goes dead after 9.31.0

pittbull, 2025-03-03:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.31.0

### ZwaveJS version

14.3.10

### Describe the bug

Using [Id-Lock 150](https://devices.zwave-js.io/?jumpTo=0x0373:0x0003:0x0001:0.0) I am seeing issues where the device randomly goes dead. I have included the requested logs and what happens in these logs is:
- Docker restarted
- Pinged the ID-Lock (node 19)
- Refreshed values (brings the device online)
- After which the node goes dead again

Not thinking 9.31.0 necessarily is the culprit, but a problem never the less.

[node_19_dump.json](https://github.com/user-attachments/files/19053162/node_19_dump.json)
[z-ui_2025-03-03.log](https://github.com/user-attachments/files/19053164/z-ui_2025-03-03.log)
[zwavejs_2025-03-03.log](https://github.com/user-attachments/files/19053163/zwavejs_2025-03-03.log)

### To Reproduce

Continuous. Not able to deduce what is the reason for the failure.

### Expected behavior

The lock to stay alive.

roflmao, 2025-03-05:
I see the same on my idlock 150 while trying to refresh usercodes;

```
<snip>
2025-03-05 10:55:09.191 INFO Z-WAVE: [Node 007] Value added: 99-0-userCode-21 =>                                                                                                                                                                                                          
2025-03-05 10:55:09.191 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-21 =>                                                                                                                                                                                                         
2025-03-05 10:55:09.928 INFO Z-WAVE: [Node 007] Value added: 99-0-userIdStatus-22 => 0                                                                                                                                                                                                    
2025-03-05 10:55:09.929 INFO Z-WAVE: [Node 007] Value added 7-99-0-userIdStatus-22 => 0                                                                                                                                                                                                   
2025-03-05 10:55:09.929 INFO Z-WAVE: [Node 007] Value added: 99-0-userCode-22 =>                                                                                                                                                                                                          
2025-03-05 10:55:09.929 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-22 =>                                                                                                                                                                                                         
2025-03-05 10:55:10.682 INFO Z-WAVE: [Node 007] Value added: 99-0-userIdStatus-23 => 0                                                                                                                                                                                                    
2025-03-05 10:55:10.682 INFO Z-WAVE: [Node 007] Value added 7-99-0-userIdStatus-23 => 0                                                                                                                                                                                                   
2025-03-05 10:55:10.683 INFO Z-WAVE: [Node 007] Value added: 99-0-userCode-23 =>                                                                                                                                                                                                          
2025-03-05 10:55:10.683 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-23 =>                                                                                                                                                                                                         
2025-03-05 10:55:11.466 INFO Z-WAVE: [Node 007] Value added: 99-0-userIdStatus-24 => 0                                                                                                                                                                                                    
2025-03-05 10:55:11.467 INFO Z-WAVE: [Node 007] Value added 7-99-0-userIdStatus-24 => 0                                                                                                                                                                                                   
2025-03-05 10:55:11.467 INFO Z-WAVE: [Node 007] Value added: 99-0-userCode-24 =>                                                                                                                                                                                                          
2025-03-05 10:55:11.467 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-24 =>                                                                                                                                                                                                         
2025-03-05 10:55:12.179 INFO Z-WAVE: [Node 007] Value added: 99-0-userIdStatus-25 => 0                                                                                                                                                                                                    
2025-03-05 10:55:12.180 INFO Z-WAVE: [Node 007] Value added 7-99-0-userIdStatus-25 => 0                                                                                                                                                                                                   
2025-03-05 10:55:12.180 INFO Z-WAVE: [Node 007] Value added: 99-0-userCode-25 =>                                                                                                                                                                                                          
2025-03-05 10:55:12.180 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-25 =>                                                                                                                                                                                                         
2025-03-05 10:55:12.935 INFO Z-WAVE: [Node 007] Value added: 99-0-userIdStatus-26 => 0                                                                                                                                                                                                    
[... cut ...]                                                        
2025-03-05 10:55:16.695 INFO Z-WAVE: [Node 007] Value added 7-99-0-userCode-31 =>                                                                                                                                                                                                         
2025-03-05 10:55:45.701 INFO Z-WAVE: [Node 007] Is now dead       
```
```
zwave-js-ui: 9.31.0.6c80945
zwave-js: 14.3.10
home id: 3759511897
home hex: 0xe0159959

AlCalzone, 2025-03-05:
@roflmao you have bad connectivity to the lock. Your controller measures background RSSI values around -85..90.
The ACKs and incoming frames on the priority route you've set are in the same ballpark (so barely distinguishable from noise):
```
routing scheme:         Priority Route
ACK RSSI:               -89 dBm
```
And even when the controller tries another route, it's not better (aborted after nearly 30 seconds):
```
2025-03-05T09:55:45.699Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            51
                                    transmit status:        NoAck, took 27650 ms
                                    repeater node IDs:      5
                                    routing attempts:       4
                                    protocol & route speed: Z-Wave, 9.6 kbit/s
                                    routing scheme:         LWR
                                    TX channel no.:         1
```

@pittbull your background noise isn't as bad, but the communication with your lock interestingly fails in the same way (aborted after 23 seconds):
```
2025-03-03T12:11:56.718Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            59
                                    transmit status:        NoAck, took 23760 ms
                                    repeater node IDs:      8
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 9.6 kbit/s
                                    routing scheme:         LWR
                                    TX channel no.:         1
2025-03-03T12:11:56.719Z CNTRLR   [Node 019] The node did not respond after 3 attempts, it is presumed dead
```

Sadly I can't be more specific, but I recommend working through https://zwave-js.github.io/zwave-js/#/troubleshooting/first-steps to try and improve the situation.