zwave-js/zwave-js #7671: Missing device configuration: Enbrighten (Jasco) Smart Fan Control 700 ( 58446 / ZWA4013 )

Nick2253, 2025-03-12:
### Checklist

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [x] It was not [merged recently](https://github.com/zwave-js/zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [x] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Enbrighten (Jasco) Smart Fan Control 700 ( 58446 / ZWA4013 )

### Manufacturer ID

0x0063

### Product Type

0x4944

### Product ID

0x3533

### Firmware Version

1.26.1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4931

### Do you have a manual?

[User-Manual-6629104.pdf](https://github.com/user-attachments/files/19203319/User-Manual-6629104.pdf)

AlCalzone, 2025-03-12:
@zwave-js-bot import config 4931 from zwa

closed 2025-05-09 by AlCalzone

kramttocs, 2025-05-28:
Thanks for getting this added. Since it's currently showing as a Light in HA, what's the correct way to get it to show as a fan? I've reinterviewed but no change.

AlCalzone, 2025-05-31:
Z-Wave has no concept of a fan for some reason, so HA will need device specific discovery to discover this as a fan instead.

Nick2253, 2025-06-25:
@AlCalzone that's not entirely correct.  Z-Wave does have a concept of a fan (device type `Fan Switch`), but such a device must be a 3-speed device.  If you want to make a device with more levels, like this device, then the only option is device type `Multilevel Switch DT`, which is basically a dimmer light switch.  This is really a problem that could be easily fixed by Z-Wave, but they haven't, so we're stuck here.

As examples, you can see that [Jasco 55259](https://products.z-wavealliance.org/products/3979?selectedFrequencyId=2) is type `Fan Switch`, but this product [Jasco 58446](https://products.z-wavealliance.org/products/4931?selectedFrequencyId=2) is type `Multilevel Switch DT`

@kramttocs To get it to show up as a fan in Home Assistant, you want to use the [Template Fan](https://www.home-assistant.io/integrations/template/#fan) in your configuration.  For example, this is how I set up my bedroom fan:

```
fan:
  - platform: template
    fans:
      master_bedroom_fan:
        unique_id: 'dcba487d-05fc-490a-9a92-0be4d00937bf'
        friendly_name: "Master Bedroom Fan"
        value_template: "{{ states('light.master_bedroom_fan') }}"
        percentage_template: "{{ ((state_attr('light.master_bedroom_fan', 'brightness') | float(0.0)) * 100/255) | round(0) |int }}"
        turn_on:
          service: light.turn_on
          entity_id: light.master_bedroom_fan
        turn_off:
          service: light.turn_off
          entity_id: light.master_bedroom_fan
        speed_count: 4
        set_percentage:
          service: light.turn_on
          entity_id: light.master_bedroom_fan
          data_template:
            brightness_pct: "{{ percentage | int }}"
```

PR #7763 (2025-04-28): feat(config): add Enbrighten 58446
Parameters based on product manual at Z-wave Alliance.

New template for Alternate Exclusion as the MENU button method seems to be what the new Jasco hardware is moving to.

Fixes #7671 and #7684