zwave-js/zwave-js #7702: Crashes with "security class is not yet known"

AlCalzone, 2025-03-31:
Originally reported in https://github.com/home-assistant/addons/issues/3952#issuecomment-2765164903
```
2025-02-24T01:29:37.479Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 166
                                    │ extensions:
                                    │ · type: SPAN  sender EI: 0x518087a72192d4dedca14bf55b962c33
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCGet]
                                      │ session id:      48
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   System
                                          notification status: 255
                                          notification event:  Heartbeat
2025-02-24T01:29:37.480Z CNTRLR   [Node 006] [Notification]
                                    type:  System
                                    event: Heartbeat
2025-02-24T01:29:37.529Z DRIVER   Dropping message with invalid payload
2025-02-24T01:29:37.529Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 166)
2025-02-24T01:29:37.674Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2025-02-24T01:29:37.675Z CNTRLR   [Node 006] Message authentication failed, cannot decode command.
2025-02-24T01:29:37.710Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2025-02-24T01:29:37.710Z CNTRLR   [Node 006] Message authentication failed, cannot decode command.
2025-02-24T01:29:38.339Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2025-02-24T01:29:38.340Z CNTRLR   [Node 006] Message authentication failed, cannot decode command.
2025-02-24T01:29:38.443Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2025-02-24T01:29:38.443Z CNTRLR   [Node 006] Message authentication failed, cannot decode command.
2025-02-24T01:29:38.522Z DRIVER   Dropping message with invalid payload
2025-02-24T01:29:38.522Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 166)
2025-02-24T01:29:38.888Z DRIVER   Dropping message with invalid payload
2025-02-24T01:29:38.888Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 166)
2025-02-24T01:29:41.507Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2025-02-24T01:29:41.507Z CNTRLR   [Node 006] Message authentication failed, cannot decode command.
2025-02-24T01:29:41.566Z DRIVER   Dropping message with invalid payload
2025-02-24T01:29:41.566Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 167)
2025-02-24T01:29:41.688Z DRIVER   Dropping message with invalid payload
2025-02-24T01:29:41.688Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 167)
2025-02-24T01:29:41.698Z DRIVER « [REQ] [SendData]
                                    callback id:            80
                                    transmit status:        NoAck, took 5740 ms
                                    routing attempts:       34
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    routing scheme:         Resort to Direct
                                    TX channel no.:         1
                                    route failed here:      11 -> 6
2025-02-24T01:29:41.703Z DRIVER » [Node 006] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      81
                                  └─[Security2CCNonceGet]
                                      sequence number: 156
2025-02-24T01:29:41.712Z DRIVER « [RES] [SendData]
                                    was sent: true
2025-02-24T01:29:42.005Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_NoSPAN
2025-02-24T01:29:42.005Z CNTRLR » [Node 006] No SPAN is established yet, cannot decode command. Requesting a non
                                  ce...
2025-02-24T01:29:50.836Z DRIVER « [REQ] [SendData]
                                    callback id:            81
                                    transmit status:        NoAck, took 9120 ms
                                    routing attempts:       33
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    routing scheme:         Explorer Frame
                                    TX channel no.:         1
                                    route failed here:      15 -> 6
2025-02-24T01:29:50.839Z DRIVER » [Node 006] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      82
                                  └─[Security2CCNonceReport]
                                      sequence number:  157
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0xc6a3086bcc4808f0fd83e435c9988163
ZWaveError: Security class for node 6 is not yet known! (ZW1402)
    at Driver.sendMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:6059:23)
    at Driver.sendCommandInternal (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:6259:28)
    at Driver.sendCommand (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:6377:29)
    at Proxy.sendReport (file:///usr/src/node_modules/@zwave-js/cc/src/cc/SupervisionCC.ts:102:20)
    at reply (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5140:9)
    at Driver.handleRequest (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5185:12)
    at Driver.handleUnsolicitedMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:4689:5)
[01:29:50] WARNING: �[33mHalt add-on�[0m
```

eric10k93, 2025-09-16:
I would be happy to.  I assume you want log level 'debug'?  I've just had it set to the default of 'info'.

From my current log level, this is all that shows up that would be relevant.  The node appears to go dead just before I get the security class error.  It's a brand new smart plug, 800LR, and it is configured as long range.  I wouldn't expect it to be having communication issues.

`2025-09-15 15:14:07.696 CNTRLR   [Node 267] The node did not respond after 1 attempts, it is presumed dead
2025-09-15 15:14:07.696 CNTRLR   [Node 267] The node is now dead.
ZWaveError: Security class for node 267 is not yet known! (ZW1402)
    at Driver.sendMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7090:23)
    at Driver.sendCommandInternal (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7286:28)
    at Driver.sendCommand (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7404:29)
    at Proxy.sendReport (file:///usr/src/node_modules/@zwave-js/cc/src/cc/SupervisionCC.ts:98:20)
    at reply (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5695:8)
    at trySupervised (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5718:12)
    at Driver.handleRequest (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5810:4)
    at Driver.handleUnsolicitedMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5201:5)
[21:14:07] WARNING: Halt add-on
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

eric10k93, 2025-09-16:
I'll see what I can do.  It's very unpredictable now, so I don't even know if I can catch it.  It was, oddly, happening constantly for a full day... I downgraded to the prior ZwaveJS release in Home Assistant and that stopped the constant crashing, but I am still seeing it occasionally, and I reverted to the current version again.

eric10k93, 2025-09-18:
[core_zwave_js_2025-09-18T18-18-42.662Z.log](https://github.com/user-attachments/files/22412764/core_zwave_js_2025-09-18T18-18-42.662Z.log)

This contains the crash and a few minutes leading up to it.

AlCalzone, 2025-09-18:
Thanks that should help narrow it down. Looking at that log, I think the error message is also wrong.

The gist of the problem:
**Device sends report while SPAN out of sync for whatever reason**
```
2025-09-18 12:10:02.648 DRIVER « [Node 267] [REQ] [BridgeApplicationCommand]
                                 │ RSSI: -100 dBm
                                 └─[Security2CCMessageEncapsulation] [INVALID]
                                     error: Security2CC_CannotDecode
2025-09-18 12:10:02.649 CNTRLR » [Node 267] Message authentication failed, cannot decode command. Requesting a 
                                 nonce...
```

**Z-Wave JS sends nonce report to re-sync**
```
2025-09-18 12:10:02.650 DRIVER » [Node 267] [REQ] [SendDataBridge]
                                 │ source node id:   1
                                 │ transmit options: 0x05
                                 │ callback id:      204
                                 └─[Security2CCNonceReport]
                                     sequence number:  213
                                     SOS:              true
                                     MOS:              false
                                     receiver entropy: 0x9ddedd447e627d110c6bfe09e56439fc
...
2025-09-18 12:10:02.662 DRIVER « [RES] [SendDataBridge]
                                   was sent: true
```

**The re-transmitted report is received, then the ACK times out**
```
2025-09-18 12:10:02.704 DRIVER « [Node 267] [REQ] [BridgeApplicationCommand]
                                 │ RSSI: -101 dBm
                                 └─[Security2CCMessageEncapsulation]
                                   │ sequence number: 131
                                   │ extensions:
                                   │ · type: SPAN  sender EI: 0xe85ece389cbcc0fb34506be0b0a77ef5
                                   │ security class:  S2_Authenticated
                                   └─[SupervisionCCGet]
                                     │ session id:      2
                                     │ request updates: false
                                     └─[MeterCCReport]
                                         meter type: Electric
                                         scale:      W
                                         rate type:  Consumed
                                         value:      127.55
                                         time delta: 0 seconds
...
2025-09-18 12:10:02.820 DRIVER « [REQ] [SendDataBridge]
                                   callback id:            204
                                   transmit status:        NoAck, took 150 ms
                                   routing attempts:       3
                                   protocol & route speed: Z-Wave Long Range, 100 kbit/s
                                   routing scheme:         Direct
                                   TX channel no.:         3
                                   TX power:               20 dBm
                                   measured noise floor:   -96 dBm

```
Not sure if the latter has an influence on this, but probably, since another NonceGet is sent:
```
2025-09-18 12:10:02.822 DRIVER » [Node 267] [REQ] [SendDataBridge]
                                 │ source node id:   1
                                 │ transmit options: 0x25
                                 │ callback id:      205
                                 └─[Security2CCNonceGet]
                                     sequence number: 215
...
2025-09-18 12:10:02.830 DRIVER « [RES] [SendDataBridge]
                                   was sent: true
[... 15 lines cut ...]
                                   callback id:            205
                                   transmit status:        NoAck, took 90 ms
                                   routing attempts:       3
                                   protocol & route speed: Z-Wave Long Range, 100 kbit/s
                                   routing scheme:         Direct
                                   TX channel no.:         3
                                   TX power:               20 dBm
                                   measured noise floor:   -96 dBm
```

**BOOM**
```
2025-09-18 12:10:02.926 CNTRLR   [Node 267] The node did not respond after 1 attempts, it is presumed dead
2025-09-18 12:10:02.926 CNTRLR   [Node 267] The node is now dead.
ZWaveError: Security class for node 267 is not yet known! (ZW1402)
    at Driver.sendMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7106:23)
    at Driver.sendCommandInternal (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7302:28)
    at Driver.sendCommand (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:7420:29)
    at Proxy.sendReport (file:///usr/src/node_modules/@zwave-js/cc/src/cc/SupervisionCC.ts:98:20)
    at reply (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5711:8)
    at trySupervised (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5734:12)
    at Driver.handleRequest (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5826:4)
    at Driver.handleUnsolicitedMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:5217:5)
[18:10:03] WARNING: [33mHalt add-on[0m
```

eric10k93, 2025-09-18:
Interesting, thanks, I will do some reading!  I was wondering if distance might be an issue, but I assumed since it was a long range chip, it would do fine (it's probably 150 ft max)... But there are a few walls and other possible barriers.

AlCalzone, 2025-09-19:
The controller is already using the maximum transmit power, and the incoming frames that are picked up show an RSSI
```
RSSI: -100 dBm
```
below the background noise near the controller:
```
measured noise floor:   -96 dBm
```

You're probably not going to be able to do anything about the signal strength of the end device, but choosing a different position for the controller has the possibility to lower the background noise levels. 
Which controller do you have? 800 series has a sensitivity around -110 dBm, which is the theoretical minimum it can report. Sometimes moving it just a few inches can be enough.

You can enable the (average) signal noise entities on the controller device and watch those for a few minutes to see if it has an effect.

eric10k93, 2025-09-22:
I have the ZWA-2 Home Assistant controller.  I just upgraded.  It's located in a mechanical room, so I wonder if maybe the electrical panel could be generating interference?  The only other device anywhere nearby with an antenna is my old HUSBZB-1, by Nortek.  I am still using the zigbee antenna, but I've completely reset the zwave chip on the device since I moved over to the ZWA-2, so I wouldn't expect it to interfere.  If there's that much background noise, I probably need to extend the ZWA-2 USB cable to another room entirely.

I did reset the zwave node that was throwing these errors and re-added and configured it as a mesh device instead of long-range.  It appears to be working better, as I haven't had a crash since doing that.  I still have a lot of pre-800 devices, so my mesh network is quite good.