zwave-js/zwave-js #7752: Missing device configuration: Fortrezz SSA missing 0x0084:0x0311:0x0109:1.9

Slyjohns, 2025-04-24:
### Checklist

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [x] It was not [merged recently](https://github.com/zwave-js/zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [x] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Frotrezz SSA1 / SSA2

### Manufacturer ID

0x0084

### Product Type

0x0311

### Product ID

0x0109

### Firmware Version

1.9

### Do you have a manual?

Missing configuration for my siren Fortrezz SSA1/2

AlCalzone, 2025-04-24:
Do you have a manual or other sources of information about the device?

Slyjohns, 2025-04-24:
i found this info in home assistant log and the URL of this unknown device :
https://devices.zwave-js.io/?jumpTo=0x0084:0x0311:0x0109:1.9

AlCalzone, 2025-04-24:
Yeah we don't know that device yet. Typically Z-Wave devices have a bunch of configuration parameters etc. which we add to the definition files. To do that, we need to know what they are.

Which is why I'm asking for a manual - those need to mention all that info and Z-Wave devices have to be sold with one.

Slyjohns, 2025-04-24:
i had this z-wave siren many years ago
it's a classique z-wave device from Fortrezz but with producttype/productid a bit different

Slyjohns, 2025-04-24:
2025-04-24T14:15:36.709Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[ManufacturerSpecificCCReport]
                                      manufacturer id: 0x84
                                      product type:    0x0311
                                      product id:      0x0109
2025-04-24T14:15:36.711Z CNTRLR « [Node 030] received response for manufacturer information:
                                    manufacturer: FortrezZ LLC (0x84)
                                    product type: 0x0311
                                    product id:   0x0109

Slyjohns, 2025-04-24:
i can confirmed you that it's a siren strobe SSA1/SSA2

Slyjohns, 2025-04-25:
Hello
Another interesting infomation, this device was well known in jeedom environment

AlCalzone, 2025-04-25:
@zwave-js-bot add fingerprint `0x0311:0x0109` to `0x0084:0x0311:0x010b:0.0`

AlCalzone, 2025-04-25:
Looks like we have some of the other fingerprints of this device, just not the one you have. Looking at the OpenZWave sources, they did group those together.

closed 2025-04-25 by AlCalzone

denix0, 2025-04-29:
@AlCalzone 

I just noticed this change in the latest update.
Funnily enough, it did not fix "Unknown product" for my FortrezZ SSA1/SSA2 strobe siren, because it is `0x0313:0x0109`, which is slightly different.

I found this manual:
https://2gig.com/wp-content/uploads/SSA1-SSA2_Instruction-Manual.pdf

And it says in there:
> Manufacturer ID: 0x0084, Product Type ID: 0x0313, Product ID: Version and Revision may vary

Should I open a separate issue, or can this be reopened/modified? Thanks!

AlCalzone, 2025-04-29:
@zwave-js-bot add fingerprint `0x0313:0x0109` to `0x0084:0x0311:0x010b:0.0`

AlCalzone, 2025-04-29:
Looks like it worked!

PR #7754 (2025-04-25): fix(config): add fingerprint `0x0311:0x0109` to "FortrezZ LLC SSA1/SSA2"
fixes: #7752