zwave-js/zwave-js #7811: VSCode extension: Recognize options that are overwritten unnecessarily

AlCalzone, 2025-05-12:
Currently, overwritten options are detected, but if they are overwritten with the exact same options, there is no complaint.

AlCalzone, 2025-07-03:
Go for it!

SChetwynd, 2025-07-04:
I've been having a look at it this morning and when I edit a config and set a `paramInformation` to import from the base template, and set the properties of the param information to overwrite the values in the template, vscode does show some errors: 

[image]

>  Is it clear what the goal here is?

Am I looking at the wrong thing here?

AlCalzone, 2025-07-04:
Almost. This is specifically about the `"options"` field. Currently, if you have them defined in the template, e.g.
```json
"options": [
  {
    "label": "Disable",
    "value": 0
  }
]
```
and define the same options in the config parameter where you import the template, it only highlights the options as being changed, but it does not recognize or warn that the options are identical to the ones in the template.