zwave-js/zwave-js #8079: Missing device configuration: Jasco Products (0x63), product type: 0x4944, product id: 0x3434

jasonbwilliams, 2025-08-17:
### Checklist

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [x] It was not [merged recently](https://github.com/zwave-js/zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [x] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Jasco Products (0x63), product type: 0x4944, product id: 0x3434

### Manufacturer ID

0x63

### Product Type

0x4944

### Product ID

0x3434

### Firmware Version

1.25

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1442?selectedFrequencyId=2

### Do you have a manual?

I have several older models of this Jasco product that are correctly identified.  In your DB, it seems these are listed as GE, but the manufacturer is not detected when included and appear as "Unknown Node".  Despite this, the device works - it just isn't labelled properly.

This is how it is seen in the logs:

2025-02-13T20:53:06.350Z CNTRLR « [Node 180] received response for manufacturer information:
manufacturer: Jasco Products (0x63)
product type: 0x4944
product id: 0x3434

[76252772.pdf](https://github.com/user-attachments/files/21824003/76252772.pdf)

apella12, 2025-08-21:
What is generally missing is the parameters.  Do you have a manual?  There are many possibilities. It is a little dicey to copy an older device.  AFAICT ZWA link is not to 4944:3434?

jasonbwilliams, 2025-08-22:
> What is generally missing is the parameters. Do you have a manual? There are many possibilities. It is a little dicey to copy an older device. AFAICT ZWA link is not to 4944:3434?

I did include a link to the manual in the post ([76252772.pdf](https://github.com/user-attachments/files/21824003/76252772.pdf)

The device works fine, it just shows up as an unknown node, with no manufacturer or model information.

apella12, 2025-08-22:
My bad.  Somehow, I missed the configuration parameters.  I'll take a look at it.

jasonbwilliams, 2025-09-30:
Can confirm the device is recognized.  Thank you!