zwave-js/zwave-js #8090: Missing device configuration: Aeotec Doorbell 6 (ZW162)

polskikrol, 2025-08-19:
### Checklist

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [x] It was not [merged recently](https://github.com/zwave-js/zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [x] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Aeotec Doorbell 6 (ZW162)

### Manufacturer ID

0x0371

### Product Type

0x0103

### Product ID

0x00a4

### Firmware Version

1.09

### Is the device listed on the Z-Wave Alliance website?

Yes, it already exists in the ZUI database but it missing this type/ID combination.

### Do you have a manual?

Here is the ZUI database item: https://devices.zwave-js.io/?jumpTo=0x0371:0x0003:0x00a2:0.0#.
Here is Aeotec upgrade information: https://aeotec.freshdesk.com/support/solutions/articles/6000252067-how-to-update-doorbell-6. Possible that IDs were modified via firmware update.

Current DB identifiers: Product Identifiers: 0x0003:0x00a2, 0x0103:0x00a2, 0x0203:0x00a2.

Need to add another one, as per the screenshot.

[image]

AlCalzone, 2025-08-20:
@zwave-js-bot add fingerprint `0x0103:0x00a4` to `0x0371:0x0003:0x00a2:0.0`

polskikrol, 2025-08-20:
[ZI01.zip](https://github.com/user-attachments/files/21900081/ZI01.zip)

Node 048 is the one in question.

AlCalzone, 2025-08-20:
There is definitely something strange with this device:
```
2025-08-20 10:49:26.830 CNTRLR   [Node 048] Version CC is not supported. Using the highest implemented version 
                                 for each CC
```
While Z-Wave JS shouldn't be skipping 95% of the interview in this case (which it does here), not supporting Version CC would be a severe violation of the specification, and fail certification almost instantly.

Missing CCs are often a temporary bug in the device firmware that typically require excluding, factory resetting and re-including the device. Can you try that first?

polskikrol, 2025-09-16:
Believe that this PR has not yet been merged. Interestingly, updated from Version: 11.2.1 to New Version: 11.3.0, and this time it picked up the correct information:

[image]

The logs were not in debug, but attached is what I have to see if that sheds any light.

[zwavejs_current.log](https://github.com/user-attachments/files/22369495/zwavejs_current.log)

polskikrol, 2025-09-17:
Vera Plus is not very good about specifying/changing/setting security keys. However, the keys can be obtained via hexdump of the /etc/cmh/keys file. I double checked that the S0 key is the same, and I assumed the S2 unauthenticated / authenticated keys the the next two hex lines in the file. In either case, re-saved the settings in ZWAVE JS and still see the decode issue for S2, and 3 failed retries for S0.

Another thought I have. How do the primary and secondary controllers exchange security keys? Vera Plus learn mode works through inclusion. So I add ZWAVE JS by placing Vera Plus into inclusion mode, and then enable to the new learn mode feature on ZWAVE JS. Does the secondary controller need to be added with S0/S2 security a) to be able to effectively communicate with other devices added in this manner or b) to even be able to exchange security keys? (Just thinking out loud, as I am surely not a subject matter expert on the protocol.)

While messing with these settings, and attempting to re-interview, still see ZWAVE JS sometimes not pick up on the correct Product / Product Code. Now see '0x0103 | Unknown product 0x00a4' again the UI.