zwave-js/zwave-js #8100: Learn Mode Support

polskikrol, 2025-08-21:
Trying to test a few things out recently, by excluding/re-including devices and adding new devices on Vera (external primary controller). My HA/ZUI instance has a connected Aeotec controller via USB. Trying to figure out how to get my Aeotec ZUI connected controller to learn from the external primary which is ID1. Supposedly, this was implemented in JS as per the following issue/feature(#7135):

No support in UI yet but you could still try using it with driver function

_Originally posted by @robertsLando in https://github.com/zwave-js/zwave-js/issues/7135#issuecomment-2497209540_
            
Additionally, quite a bit of work was completed for UI to support learn mode via PR 4097:  https://github.com/zwave-js/zwave-js-ui/pull/4097. This has been open since January of this year.

Would be happy to try the 'driver function' method but the only template I can find is for cloning nodes which isnt exactly the same thing. Could any assistance be provided? Perhaps a 'driver function' template written or the PR 4097 completed?

Happy to test but just need some guidance.

polskikrol, 2025-08-21:
Opened this via a link in Github to 'reply and open an issue', so didnt have the nice template handy to complete :/ Hopefully apart from driver logs (not yet pertinent for this issue), the following covers the rest of the usually needed bits of information:

zwave-js-ui: 11.1.0.c235c6c
zwave-js: 15.10.0

Primary (external): Vera 1.7.5186 (7.31)
Secondary (USB connected): Aeotec FW: v7.21.6

marcus-j-davies, 2025-08-24:
I don't use UI - but do use (and testing) the driver in Secondary Mode, I am not sure how this translates to UI, but these are the steps.

Add a call back - to the ZwaveJS Driver settings.

```js
joinNetworkUserCallbacks: {
      showDSK:fucntion(DSK){ console.log(DSK) }, /* Not sure how this is handled in UI, so just putting in console */
      done: function(){}
},
```

On the primary, put your controller into Inclusion  Mode - the primary controller will want to know the DSK, and the callback will be trigged once it requests it.

Now  put driver into Learn Mode

```js
driver.controller.beginJoiningNetwork()
```

There is a driver event, once the handshake has happened : `network joined`, and from here you can fetch the nodes on the network, that Zwave JS has just joined.

It will require work on UI - but the driver itself (after my testing) is pretty solid here.
You will need to manually interview Nodes, and update Life Lines, if you want ZwaveJS to be updated with value changes.

Hope it helps.

polskikrol, 2025-09-10:
This issue can be closed as https://github.com/zwave-js/zwave-js-ui/pull/4097 was merged. But running into some issues, please advise if I should open a separate bug against zwave-js or zwave-js-ui instead of commenting on the PR.

closed 2025-09-10 by robertsLando