zwave-js/zwave-js #8112: Missing device configuration: <enter manufacturer and device name>

itssomthing, 2025-08-27:
### Checklist

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [x] It was not [merged recently](https://github.com/zwave-js/zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [x] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

0x014f:0x4744:0x3530

### Manufacturer ID

0x014f

### Product Type

0x4744

### Product ID

0x3530

### Firmware Version

0.0

### Do you have a manual?

https://www.alarmgrid.com/documents/gocontrol-gd00z-8-gc-install-guide-dated-2020-rev-b?

itssomthing, 2025-08-28:
https://www.alarmgrid.com/documents/linear-gd00z-4-installation-manual?

AlCalzone, 2025-09-08:
This device exists in our database: https://devices.zwave-js.io/?jumpTo=0x014f:0x4744:0x3030:0.0